pynvme/pynvme #11: Assertion error attaching to NVMe Controller

brennanwatt, 2019-06-10:
I am hitting assertion error in driver.c when trying to run scripts/demo_test.py. Here is the output:

`[root@localhost pynvme]# python3 -m pytest scripts/bw_test.py::test_trim_basic --pciaddr=01:00.0 -s
Starting SPDK v19.01 / DPDK 18.08.0 initialization...
[ DPDK EAL parameters: pynvme -c 0x8 -m 512 --base-virtaddr=0x200000000000 --file-prefix=spdk0 --proc-type=auto ]
EAL: Detected 12 lcore(s)
EAL: Detected 1 NUMA nodes
EAL: Auto-detected process type: PRIMARY
EAL: Multi-process socket /var/run/dpdk/spdk0/mp_socket
EAL: No free hugepages reported in hugepages-1048576kB
============================================================================================= test session starts =============================================================================================
platform linux -- Python 3.6.3, pytest-4.6.2, py-1.8.0, pluggy-0.12.0 -- /opt/rh/rh-python36/root/usr/bin/python3
cachedir: .pytest_cache
rootdir: /home/ssdfw/brwat/pynvme, inifile: pytest.ini
collected 1 item

scripts/bw_test.py::test_trim_basic
----------------------------------------------------------------------------------------------- live log setup ------------------------------------------------------------------------------------------------
[2019-06-10 11:24:41.054] INFO pciaddr(19): running tests on DUT 01:00.0
driver.c: 637:probe_cb: *INFO*: Attaching to NVMe Controller at 0000:01:00.0
python3: driver.c:539: intc_init: Assertion `(table_offset&0x7) == 0' failed.
Aborted
`

It looks like I am failing attaching to the NVMe controller because the lower 3 bits of table_offset are not zero. I have confirmed I have the correct BFD.

Any ideas?

Note: I am trying to get things running on CentOS 7. I also tweaked the demo_test.py script to stop trying to import k2c because it was failing for the following:
`E   ModuleNotFoundError: No module named 'pytemperature'`

cranechu, 2019-06-10:
Hi, Brennan, 

This assert means we expecting MSIx table located in BAR0, but your device's MSIx table is in BAR4. Can you let us know which device are you using? Is it available in the market? We can try to reproduce it by ourselves. 

Or, we will create a new branch, and need your support to test and debug it with us. 

Thanks, 
-Crane

cranechu, 2019-06-11:
pynvme extended MSIx function from SPDK codebase with a quick (and dirty) solution, which assumes the MSIx table is located in BAR0. It is just true on our tested drives, but it is not always true. We need to map the MSIx table from the correct BAR with API: spdk_pci_device_map_bar(). 

Also consider the same issue with CMB and boot partition.

cranechu, 2019-07-16:
work in progress: https://github.com/cranechu/pynvme/tree/intr_fix

brennanwatt, 2019-09-11:
Confirmed that the latest master code resolves the issue and can work with drives that have MSIx in BAR4

closed 2019-09-11 by brennanwatt