pynvme/pynvme #20: Can pynvme support MSI interrupts mode?

oulaiwanle, 2019-07-03:
Hi:
  I am trying to use pynvme to test our nvme ssd .
  when i use command :`nvme.Controller(b'01:00.0’）`,there is an  assertion error :
  driver.c: 637:probe_cb: *INFO*:Attaching to NVMe Controller at 0000:01:00.0
  python3: driver.c:517: intc_find_msix: Assertion 'next offset !=0' failed.
  Aborted (core dumped)

  Our Nvme SSD suppurts MSI but does not support MSI-X .
  Can pynvme support MSI interrupts mode ?
 
Thanks.

cranechu, 2019-07-04:
Hi, 

It's really a different device. Thanks for your issue reporting, and we will consider this request. 
But before that, you can try this branch: br_msix_disabled. It just disables MSIx, so you will not be blocked by this assert. 

Thanks, 
-Crane

RogerWYQ, 2019-10-14:
> Hi:
> I am trying to use pynvme to test our nvme ssd .
> when i use command :`nvme.Controller(b'01:00.0’）`,there is an assertion error :
> driver.c: 637:probe_cb: _INFO_:Attaching to NVMe Controller at [...]

Hi oulaiwanle
MSI intr mode is supported in latest pynvme. So would you please try it and test your device. If you have any additional issue, please let us know.

closed 2019-10-31 by cranechu