pynvme/pynvme #23: make test report error “failed to create the controller”

chens54, 2019-07-31:
During pynvme installation, when I tried to execute "make test", error message displays as following:
`-------------------------------- live log setup --------------------------------
[2019-08-01 12:47:15.155] INFO pciaddr(19): running tests on DUT 04:00.0
EAL: Cannot write command to PCI config space!
EAL: Cannot set up bus mastering!
EAL: Driver cannot attach the device (0000:04:00.0)
created ctrlr: 0
ERROR

==================================== ERRORS ====================================
_____________________ ERROR at setup of test_create_device _____________________

pciaddr = '04:00.0'

    @pytest.fixture(scope="session")
    def nvme0(pciaddr):
>       ret = d.Controller(pciaddr.encode('utf-8'))

conftest.py:38:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
driver_wrap.pyx:821: in nvme.Controller.__cinit__
    self._create()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

>   raise NvmeEnumerateError(f"fail to create the controller")
E   nvme.NvmeEnumerateError: fail to create the controller

driver_wrap.pyx:836: NvmeEnumerateError
------------------------------ Captured log setup ------------------------------
INFO     root:conftest.py:19 running tests on DUT 04:00.0
=========================== short test summary info ============================
ERROR driver_test.py::test_create_device - nvme.NvmeEnumerateError: fail to c...
SKIPPED [1] driver_test.py:47: nvme over tcp
====================== 1 skipped, 1 error in 0.14 seconds ======================
Makefile:100: recipe for target 'pytest' failed
cat test.log | grep "434 passed, 2 skipped" || exit -1
/bin/sh: 1: exit: Illegal number: -1
Makefile:103: recipe for target 'test' failed
make: *** [test] Error 2
`

Could you help to figure out this issue for me?
The whole log is 
[temp.txt](https://github.com/cranechu/pynvme/files/3450684/temp.txt)
If you have any concern, please contact me.
Looking forward to you response, thanks.

RogerWYQ, 2019-08-01:
Hi Chens54
would you please try setpci command to access the pcie configuration space? if the result is the same as you mentioned above. Please check the secure boot setting in your bios configuration. I have met the issue before, because the secure boot setting of my platform is enabled, which will enable memory lockdown mode to prohibit accessing kernel memory.

chens54, 2019-08-02:
May I know how to use setpci command? I'm not familiar with how to use it.

cranechu, 2019-08-05:
Hi, chens54, 

"make test" only passes with specific NVMe devices. We use it as the acceptance test of pynvme. 

If you need to run tests with your own SSD device, you may need to modify some test cases in  driver_test.py, or create your own test scripts under the directory "scripts". 

Would you mind if upload your failure screenshot here, as well as the test.log? So we can get more information for further debug.

chens54, 2019-08-13:
I fixed this issue by
1. Disable security boot in BIOS
2. using sudo to launch pycharm.

Thanks @cranechu

cranechu, 2019-08-15:
> I fixed this issue by
> 
> 1. Disable security boot in BIOS
> 2. using sudo to launch pycharm.

Great! Thanks for sharing your status. 
Pycharm is beautiful. But if you could try vscode, you may find its pynvme plugin is useful for debugging test scripts. 

Thanks, 
-Crane

closed 2019-08-15 by cranechu