pynvme/pynvme #29: Question about reset during io_worker

catogts, 2019-10-10:
Hi. 
Thank you make for superior nvme test tool
I test nvme reset during ioworker using this tool on intel desktop and dell server
But two system show different result.
Desktop issue segment fault after abort pending io but dell server continue test.
I want to know that available reset test during iowoker test
Version 1.3, 1.4 same result
Thank you

cranechu, 2019-10-11:
Hi, catogts, 

It is not supported to reset during ioworker. Another similar operation is power cycle. For now, we have to complete all outstanding commands and ioworkers before making a reset or power cycle. 

Thanks, 
-Crane

catogts, 2019-10-14:
Thank you~
many nvme issue occur in iowok during reset at ssd development
i wish avaliable this func in future pynvme

cranechu, 2019-10-14:
Yes, exactly! We will make some efforts on this feature. 

Thanks, 
-Crane

cranechu, 2019-10-15:
Reopen. We will track the progress in this page.

cranechu, 2020-01-10:
Before reset and power off, notify all ioworkers to be terminated immediately. 
Here is an example:

``` python
def test_ioworker_power_cycle_async(nvme0n1, subsystem):
    for i in range(2):
        start_time = time.time()
        with nvme0n1.ioworker(io_size=8, time=100), \
             nvme0n1.ioworker(io_size=8, time=100), \
             nvme0n1.ioworker(io_size=8, time=100):
            time.sleep(5)
            subsystem.power_cycle(10)
        # terminated by power cycle
        assert time.time()-start_time < 25

    with nvme0n1.ioworker(io_size=8, time=10):
        pass
    subsystem.power_cycle(10)
```

closed 2020-01-10 by cranechu