pynvme/pynvme #30: spdk v20.01

cranechu, 2019-10-14:


cranechu, 2019-11-01:
Added new error handling and reporting functionality. This includes several
new API functions to facilitate applications recovering when a qpair or
controller fails.

spdk_nvme_ctrlr_reconnect_io_qpair - Reconnects a failed I/O qpair.
spdk_nvme_ctrlr_set_trid - Sets the trid of an existing controller. Can be used to
change the trid for failover cases.
spdk_nvme_ctrlr_is_failed - Returns the failed state of a controller.
spdk_nvme_ctrlr_fail - Forces a controller into a failed state.

Modified the return behavior of several API functions to better indicate to
applications when a qpair is failed. This list of functions includes:

spdk_nvme_qpair_process_completions
spdk_nvme_ns_cmd_*
spdk_nvme_ctrlr_process_admin_completions
spdk_nvme_ctrlr_cmd_*

catogts, 2019-11-02:
Do you have plan to apply spdk 19.10 ?

cranechu, 2019-11-03:
We have applied spdk 19.10 in pynvme_1.5, the next upcoming release.

closed 2019-11-03 by cranechu