pynvme/pynvme #31: Have question some device problem in visual studio  code pynvme plug in

catogts, 2019-10-19:
Hi

Some enterprise essd have problem in visual studio code plug in queue window

These device print some hex code at device active (nvme0=d.Contoller(...)) unlike normal device
(After MSIX address print)
Device operation no problem.  But not show at queue pair status window

How can i see all log at this time?

Thank you

cranechu, 2019-10-21:
Hi, 

Could you please upload or email us a screenshot and the test.log generated in the test? We can learn more details of your situations. 

Thanks,
-Crane

catogts, 2019-10-22:
Hi.
no test.log. 
I only run  'sudo python3' and 'import nvme as d' , ' nvme0=d.Controller(b'0c:00.0')
result is
INFO intr_mgt.c(176) msix_intc_int: vextor_num 32
INFO intr_mgt.c(183) msix_intc_int: msix table addr 201000800000
INFO intr_mgt.c(192) msix_intc_int: msix table addr 2 0x201000800000
INFO intr_mgt.c(195) msix_intc_int: msix table addr 201000803000
00000000 18 62 32 61 13 5b 45 5f b9 4e c5 34 66 1b 9e 69 .b2a.[E_.N.4f..i
INFO driver.c(601) nvme_init: found device: 0000:0c:00.0, 0x2000201e6140
>>

other device not print abnormal hex code. only this device occur this issue

operation no problem.

cranechu, 2019-10-22:
Thanks for reporting the detail finding. We will investigate it.

cranechu, 2019-10-22:
> Controller Attributes (**CTRATT**): Bit 0 if set to ‘1’, then the controller supports a 128-bit Host Identifier. 

Your enterprise SSD likely supports this feature, and so SPDK generates the ID, and print it in the log. 
So, it is just normal output.

catogts, 2019-10-22:
Thank you
If ctratt is set then unavailable visual code plugin?

cranechu, 2019-10-22:
Could you please upload a screenshot? I may need your help to reproduce and test this issue. I dont have the device to reproduce it. 

Thanks, 
-Crane

catogts, 2019-10-23:
Sorry~ can not take screenshot because company security reason
normal sample all show Qpair Status after nvme0=d.Controller(b'xx:xx.0')
but some sample not show qpair status after nvme0=d.Controller(b'xx:xx.0').
only progress bar blink upper "PYNVME QPAIR" status window name
other enterprise sample and consumer ssd sample no problem.
how can i see driver level detail log?

catogts, 2019-10-24:
Thank you
i tested debug driver. 
i compared issue devices to normal device debug message. no difference devices except one thing
(i disabled 128bit Host ID to issue device. but still remain issue)
issues device have special unicode in fw name place last character (0xfffd)
each device print DEBUG driver.c(510) attach_cb: attached device...
normal device : 0000:be:00.0:  device name 1.0.4  , 1 namespaces, pid xxx
issue device : 0000:bc:00.0: device name 1.0.4.�, 1 namespaces, pid xxx
Can certain Unicode cause problems?

cranechu, 2019-10-24:
Hi, 

Can you share with me the debug log of the device with issues? Maybe I can find some more interesting points. 

For those Unicode device name, it violates the spec, but I don't think it would cause any problem. 
AQA set before base address: it is also fine, only if those registers are written before CC.EN

In you first message, you mentioned:
> Device operation no problem. But not show at queue pair status window
So, the devices with symptoms of unicode/nvme-mi/AQA still can work as usual, but only vscode plugin does not display qpairs of them?  

You can send me email if it is more efficient for you. 

Thanks, 
-Crane

cranechu, 2019-10-24:
Here is my test.log file when I run the commands below with branch issue_31:
```shell
make setup
make test TESTS=scripts/demo_test.py::test_debug
```
[test.log](https://github.com/cranechu/pynvme/files/3767042/test.log)

catogts, 2019-10-25:
ok i see seems like this log
but can  not find difference  point
i dont know why can not display vsc plugin queue starus

cranechu, 2019-10-25:
If you can send me the log file, I may find more details.

catogts, 2019-10-29:
Ok i will try logfile to you

cranechu, 2019-10-31:
Hi, Catogts, 

We just meet a similar issue recently that vscode cannot display controllers and qpairs. We have fixed it in the branch: https://github.com/cranechu/pynvme/tree/pynvme_1.5

Could you please repeat the test with your SSD?

catogts, 2019-10-31:
Thank you!

Works great

cranechu, 2019-10-31:
Perfect!