pynvme/pynvme #33: How to test multiple device?

catogts, 2019-10-30:
Hi

I tried test multiple device using multi linux account

but only 1 pynvme operate  available 

how can i test multiple device at onece

i want to test mass devices

cranechu, 2019-10-31:
Here is an example for your reference: 

```python
def test_multiple_controllers_and_namespaces():
    # address list of the devices to test
    addr_list = [b'3a:00.0', b'10.24.48.17']

    nvme_list = [d.Controller(a) for a in addr_list]
    ns_list = [d.Namespace(n) for n in nvme_list]

    # operations on multiple controllers
    for nvme in nvme_list:
        logging.info("device: %s" % nvme.id_data(63, 24, str))

    # multiple namespaces and ioworkers
    ioworkers = {}
    for ns in ns_list:
        a = ns.ioworker(io_size=8, lba_align=8,
                        region_start=0, region_end=256*1024*8, # 1GB space
                        lba_random=False, qdepth=16,
                        read_percentage=100, time=10).start()
        ioworkers[ns] = a

    # test results
    for ns in ioworkers:
        r = ioworkers[ns].close()
        io_total = (r.io_count_read+r.io_count_write)
        logging.info("capacity: %u, IOPS: %.3fK" %
                     (ns.id_data(7, 0), io_total/10000))
```

catogts, 2019-10-31:
Thank you very much

very useful examples

closed 2019-10-31 by cranechu