pynvme/pynvme #34: Availble use ptype, pvalue in iowoker read?

catogts, 2019-10-31:
Hi

I tested write(using ioworker) -> sanitize -> read(using ioworker) sequence 

But data mismatch failure occurred during a read sequence 
(seems like verify function not recognize sanitize operation)

Then i tried read iowoker using ptype=0, pvalue=0 for sanitized area data

But still failed verified data because read data 0x00

How can i resolve this situation?

Thank you

cranechu, 2019-11-01:
Hi, 

We have a similar test cases in driver_test.py, called test_sanitize_basic(). Here is a pipeline in gitlab for this script, and it passes: https://gitlab.com/cranechu/pynvme/-/jobs/339083470
We can double check your script if necessary. 

For ptype/pvalue, they are only used for write data pattern. So, no impact on read. 

Thanks, 
-Crane

catogts, 2019-11-01:
i tested upper script but last io work iops very low becae almost lba mismatch fail 
if i do ctrl+brake than i can see lba mismatch fail log at ioworker read timing
could you check last read ioworker result?

cranechu, 2019-11-01:
After  sanitize, all LBAs are unmapped. So pynvme will not do data verification. Could you please send me the test.log file? And, "sanitize mode is 0x11" ==> what does it mean?

catogts, 2019-11-01:
Ok~ i understand!
sanitize mode 0x11 is my mistake
i confuse capability and set register

closed 2019-11-01 by catogts