pynvme/pynvme #35: Subsystem Reset issue ERROR nvme_ctrlr.c

catogts, 2019-11-05:
Hi crane

I tested pynvme at 2 platform (1 desktop, 1 server)

but both system can not operate subsystem reset

pynvme occur error below
--desktop log--
ERROR nvme_ctrlr.c(728) nvme_ctrlr_shutdown: get_csys() failed
python3: nvme_ctrl.c:728: nvme_ctrlr_shutdown: Assertion 'false' failed
--server log--
INFO Driver.c(498) probe_cb: Attaching to NVMe Controller at 0000:xx:xx.x
ERROR nvme_pcie.c(488) nvme_pcie_ctrlr_map_cmb: get registers failed
python3: nvme_pcie.c:488: nvme_pcie_ctrlr_map_cmb: Assertion 'false' failed.

any sample can not subsystem reset.

i use master brance

cranechu, 2019-11-06:
From NVMe Spec: 

> 10.5 Controller Fatal Status Condition
> 
> Performing an NVM Subsystem Reset (NSSR) may cause PCI Express links to go down as part of resetting the NVM subsystem. Host software may have undesirable effects related to PCI Express links going down (e.g., some host operating systems or hypervisors may crash). 

What you observed is just because PCIe links down.

cranechu, 2019-11-06:
We also fill confuse about this operation. Some drives fail as the same link down issue, but some drives do pass. Need more investigation.

catogts, 2019-11-07:
Thank you
i will try more sample and system
availble pcie link setup in driver level?

cranechu, 2019-11-08:
> availble pcie link setup in driver level?

You can try Pcie.reset()

closed 2019-11-28 by cranechu