pynvme/pynvme #38: issue about EAL errors

catogts, 2019-11-20:
hi crane

sometimes i see below erros at close ioworkers

EAL: failed to send to (/var/run/dpdk/spdk0/mp_socket_21381_a0284f9ff2a) due to No such file or directory
EAL: Fail to send request /var/run/dpdk/spdk0/mp_socket_21831_a0284f97ff2a:mp_malloc_sync
EAL: Fail to handle messages: mp_malloc_request
EAL: Wrong request ID
EAL: Request timed out
EAL: failed to send to (/var/run/dpdk/spdk0/mp_socket) due to Bad file descriptor
EAL: recvmsg failed, Bad file descriptor

i use test_read_multiple_devices_50hr in marathon_test.py

System is 2 xeon CPU, 80core, 384GB, target nvme device is 8TB 4ea

not happen always this issue. but this issue stop the test in vscode

can multi processor cause problem?

cranechu, 2019-11-20:
Hi, 

I don't have dual socket platform, so cannot reproduce it. Could you please upload the log file? Or a part of the log? Thanks! 

I am also running the same script in single xeon CPU. 

-Crane

cranechu, 2019-11-21:
Hi, Catogts, 

The same test passes in the single xeon CPU platform. Could you please provide more information for analysis? 

Thanks, 
-Crane

catogts, 2019-11-21:
I disabled NUMA on Linux kernel (Centos 7)

no more same issue.

maybe SPDK memory fregment between multi processor

cranechu, 2019-11-28:
recommend to disable numa

closed 2019-11-28 by cranechu