pynvme/pynvme #40: Some Question and Some Issue about reset

catogts, 2019-12-06:
Hi crane

I have some question about multiple device test and some issue about reset

first, issue about reset

pcie reset and subsystem reset is not working correctly in my company pc(intel skylake) and server (dell r740xd) and my home pc (amd ryzen system)

all system report error in nvme_ctrlr.c (line 702 and line 728)  during pcie reset

seems like systems can not read register. 

after i comment out these lines, pcie reset and subsystem working correctly

second, question about multiple device test

Currently, one script can run a different workload on each ssd, but if one problem occurs a one ssd, all of them should stop

Is there a way to run the test(different script) on each ssd individually?

thank you

cranechu, 2019-12-07:
Hi, 

Thanks very much for your questions. For the first issue, I also found it and fixed it in v1.6. Could you please try it again with v1.6? 

For the second one, I have not tested in this way, but it is a very interesting topic. Could you please re-submit the second issue in another seperated issue? So we can trace the progress better. 

My first idea is the container. Let us make a try to run pynvme tests in conrainers? :)

catogts, 2019-12-09:
ok~ i will open seperated issue about sencond one

and i checked reset issue cleared from ver 1.6 

thank you

closed 2019-12-09 by catogts