pynvme/pynvme #42: Suggest test method

catogts, 2019-12-14:
I suggest test mode for Individual test for each device

Some nvme test systems use a separate driver for each device

It would be nice if pynvme had a way to run different tests individually for each device.

cranechu, 2020-01-23:
kvm + iommu + pci passthrough: run pynvme in virtual machine with centos7

closed 2020-01-23 by cranechu

catogts, 2020-01-23:
> kvm + iommu + pci passthrough: run pynvme in virtual machine with centos7

Hi Crane 

Do I need to compile spdk with vfio enabled?

cranechu, 2020-01-23:
Not required. pynvme still works with uio within the virtual environment. However, iommu (or VT-d) should be enabled in host machine, in order to passthrough PCIe devices.

catogts, 2020-01-24:
> Not required. pynvme still works with uio within the virtual environment. However, iommu (or VT-d) should be enabled in host machine, in order to passthrough PCIe devices.

I tested kvm+iommu+pcipassthrough

iowork and cmd base test is good

but not work power cycle or Subsystem Reset. i think kvm not support pci passthrough hotplug

cranechu, 2020-01-25:
> I tested kvm+iommu+pcipassthrough
>
> iowork and cmd base test is good
>
> but not work power cycle or Subsystem Reset. i think kvm not support pci passthrough hotplug

Thanks for sharing the progress! 
For subsystem power_cycle and reset, we need more investigations. 
Can you email me a test.log file? 

Thanks, 
-Crane

catogts, 2020-02-21:
hi crane

i tested pynvme on qemu environment

ioworkload test is good. but not operate subsystem reset, pcie reset, powercycle test

host iommu can not remap irq during power cycle and pcie reset because qemu hold pass though pcie nvme

host iommu rmap irq after guest pc off

multiple spdk driver load unavaliable?