pynvme/pynvme #43: Issue about multiple ioworker

catogts, 2020-01-07:
Hi crane

I have some issue about multiple iowoker

If i set multi iowoker to different io align and lba start. All ioworker have same workload

Iowoker a : io size :1 , io align :4 , start lba 0
Iowoker b: io size :1 , io align :4 , start lba 1
Iowoker c : io size :1 , io align :4 , start lba 2
Iowoker d : io size :1 , io align :4 , start lba 3

I expect 

Iowoker a : 0,4,8,c
Iowoker b : 1,5,9,d
Iowoker c : 2,6,a,e
Iowoker d : 3,7,b,f

But all iowoker 0,4,8,c....

Please check this issue

cranechu, 2020-01-07:
Hi, 

What you observed is just current design: all LBAs are aligned to 4. 
You expected LBA pattern cannot be implemented by current ioworker. If you really need this IO pattern, I can add one more parameter to ioworker: io_stride for sequential read/write works only. 

Thanks, 
-Crane

catogts, 2020-01-08:
thank you. crane

I've seen some enterprise workloads send only odd, even, or some pattern LBA.

i think need support this

cranechu, 2020-01-10:
Added parameter lba_step to Namespace.ioworker(). So, you can implement your test as below:
Iowoker a : io size :1 , io align :1 , start lba 0, lba_step 4
Iowoker b: io size :1 , io align :1 , start lba 1, lba_step 4
Iowoker c : io size :1 , io align :1 , start lba 2, lba_step 4
Iowoker d : io size :1 , io align :1 , start lba 3, lba_step 4

Here are more examples: 
https://github.com/pynvme/pynvme/blob/d5ca45850fd3424e9580eb1a2f91cea2a3127c92/driver_test.py#L1431-L1477

closed 2020-01-10 by cranechu