pynvme/pynvme #46: Issue about ioworker stability

catogts, 2020-01-18:
Hi Crane

I wonder if nvme capacity can affect ioworker stability.

i test 16T nvme drive on dell xeon sever(384GB). about simply bellow code

-----------------------------------------------------------------------------
nvme0=d.Controller(b'0a:00.0')
nvme1=d.Controller(b'0b:00.0')

nvme0n1=d.Namespace(nvme0)
nvme1n1=d.Namespace(nvme1)

sub0=d.Subsystem(nvme0)
sub1=d.Subsystem(nvme1)

for loop in range(9999):
    iowokers_a=nvme0n1.iowoker(io_size=random.randrange(1,64), read_percentage=0, qdepth=1023,time=30).start()
    iowokers_b=nvme1n1.iowoker(io_size=random.randrange(1,64), read_percentage=100, qdepth=1023,time=30).start()

    time.sleep(random.rangrange(5,15))

    sub0.power_cycle()
    sub1.power_cycle()

    ioworkers_a.close()
    ioworkers_b.close()
------------------------------------------------------------------------------
(I replaced the quanch of the power cycle code with labjack. works great)

When I tested two 1T nvme with the above code, no problem occurred(about 1000 cycle) but when I test two 16T nvme, I often get a segment fault or python abort in the python multi thread module

Do I need another fix to test large nvme?

thank you

catogts, 2020-01-20:
some fail message is
def spawnv_passfds(path, arfs,passfds):
 import _posixsubprocess
 passfds = tuple(sorted(map(int, passfds)))
   errpip_read, errpipe_write=os.pipe()
 OSError: [Errno 24] Too many open files

catogts, 2020-01-20:
> some fail message is
> def spawnv_passfds(path, arfs,passfds):
> import _posixsubprocess
> passfds = tuple(sorted(map(int, passfds)))
> errpip_read, errpipe_write=os.pipe()
> OSError: [Errno 24] Too many open files

this error cleared by linux number of open files option

but not cleared below error

/usr/lib64/python3.6/socket.py
/home/ssd/.cscode/extensions/ms-python.python-2020.1.58038/pythonFiles/liub/python/old_pyvsd/pyvsd/messageing.py, line 70 in read_line
/usr/lib64/python3.6/socket.py
/home/ssd/.cscode/extensions/ms-python.python-2020.1.58038/pythonFiles/liub/python/old_pyvsd/pyvsd/messageing.py, line 88 in read_json
/usr/lib64/python3.6/socket.py
/home/ssd/.cscode/extensions/ms-python.python-2020.1.58038/pythonFiles/liub/python/old_pyvsd/pyvsd/messageing.py, line 434 in _process_incoming_messages

is this related visual stdio code ?

cranechu, 2020-01-20:
Hi, Catogts, 

Did you run the test in VSCode? Could you please re-run the test in command line as blow:
``` shell
   make test TESTS=scripts/your_test_file.py"
```
If possible, please send the generated test.log file to cranechu@gmail.com, I can collect more information to debug. I don't have 16T NVMe drive, so cannot reproduce your issue here. 

Thanks, 
-Crane

cranechu, 2020-01-20:
Can you show me the web page of the labjack you are using? Thanks!

catogts, 2020-01-20:
> Can you show me the web page of the labjack you are using? Thanks!

https://labjack.com/products/u3

i use this model and customize pcb.

labjack send only control signal (set.FIOState)

catogts, 2020-01-20:
> Hi, Catogts,
> 
> Did you run the test in VSCode? Could you please re-run the test in command line as blow:
> 
> ```shell
>    make test TESTS=scripts/your_test_file.py
> ```
> 
> If possible, please send the generated test.log file to [cranechu@gmail.com](mailto:cranechu@gmail.com), I can collect more information to debug. I don't have 16T NVMe drive, so cannot reproduce your issue here.
> 
> Thanks,
> -Crane

ok i'll try this

catogts, 2020-01-21:
make test TESTS=scripts/your_test_file.py

not happen error when using upper method (about 12hrs)

cranechu, 2020-01-21:
> make test TESTS=scripts/your_test_file.py
> 
> not happen error when using upper method (about 12hrs)

Great! 
Personally, I develop and debug scripts in VSCode or Emacs, and execute them in command line environment. The command line should be more stable. 

Thanks, 
-Crane

closed 2020-01-21 by cranechu