pynvme/pynvme #48: ERROR nvme_ctrlr.c(2489) nvme_ctrlr_process_init:                               Initialization timed out in state 3

GongQin721, 2020-03-02:
你好，我想请问下我在中断运行一个脚本的时候，手动中断后，再次运行其他脚本，就会出现error:
 INFO driver.c(498) probe_cb: Attaching to NVMe Controller at 0000:01:00.0
[2020-03-02 11:19:07.377150] INFO intr_mgt.c(174) msix_intc_init: vector_num 32
[2020-03-02 11:19:57.234873] ERROR nvme_ctrlr.c(2489) nvme_ctrlr_process_init:                               Initialization timed out in state 3
python3: nvme_ctrlr.c:2489: nvme_ctrlr_process_init: Assertion `0' failed.
Fatal Python error: Aborted
我想请问下这个可能是什么原因导致的呢，我看百度上说的是SPDK模式的盘已经有进程绑定了，当前进程无法attach 到SPDK模式的盘。kill调所有使用SPDK模式盘的进程，重新拉起程序。但是我看没有这个进程在运行啊？

cranechu, 2020-03-02:
Yes, you need to kill all processes running SPDK and pynvme. Because pynvme tests are organized by pytest, so you can search by 'pytest' to get the pid as below:
```shell
ps -eo pid,args | grep pytest
```

GongQin721, 2020-03-02:
嗯嗯，多谢了

GongQin721, 2020-03-02:
你好，我输入你发的那个命令，kill的时候发现没有这样的进程，但是运行脚本还是出现超时

closed 2020-03-09 by cranechu