pynvme/pynvme #49: nvme.NvmeEnumerateError: fail to create the controller

seanmaster, 2020-03-09:
Hi cranechu,
I am interested about nvme test, so I clone the latest pynvme to learn.
When I test about power cycle test, I got an error as title
Below is a detail about log.

=================================== FAILURES ===================================
___________________________ test_powercycle_by_sleep ___________________________

subsystem = <nvme.Subsystem object at 0x7fffe818c438>

    def test_powercycle_by_sleep(subsystem):
        # sleep system for 10 seconds, to make DUT power off and on
>       subsystem.power_cycle()

scripts/test_utilities.py:23: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
driver_wrap.pyx:382: in nvme.Subsystem.power_cycle
    ???
driver_wrap.pyx:638: in nvme.Controller._reinit
    ???
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

>   ???
E   nvme.NvmeEnumerateError: fail to create the controller

driver_wrap.pyx:664: NvmeEnumerateError
------------------------------ Captured log setup ------------------------------
INFO     root:conftest.py:55 running tests on DUT 0000:0e:00.0
INFO     root:conftest.py:68 setup random seed: 0xd6c32b5b
------------------------------ Captured log call -------------------------------
INFO     root:test_utilities.py:23 power off nvme device for 10 seconds
INFO     root:test_utilities.py:23 power is back
---------------------------- Captured log teardown -----------------------------
INFO     root:conftest.py:70 test duration: 42.462 sec
=========================== short test summary info ============================
FAILED scripts/test_utilities.py::test_powercycle_by_sleep - nvme.NvmeEnumera...
============================== 1 failed in 42.56s ==============================
make[1]: Leaving directory `/home/sean/pynvme'

How I can fix it? Very thank you~~

cranechu, 2020-03-10:
Hi, seanmaster

Thanks for using pynvme. 
Could you please run the command
``` shell
make info
```
and attach the output? I need some information about your testing platform. 

Thanks,
-Crane

seanmaster, 2020-03-10:
Hi cranechu,
below is my platform info.

[sean@seanlinux pynvme]$ make info
sudo ./spdk/scripts/setup.sh status
Hugepages
node     hugesize     free /  total
node0   1048576kB        0 /      0
node0      2048kB     2450 /   2450

NVMe devices
BDF		Vendor	Device	NUMA	Driver		Device name
0000:0e:00.0	126f	2262	0	uio_pci_generic		-

I/OAT DMA
BDF		Vendor	Device	NUMA	Driver

virtio
BDF		Vendor	Device	NUMA	Driver		Device name
VMD
BDF		Numa Node	Driver Name
sudo cat /proc/meminfo
MemTotal:        8035464 kB
MemFree:          917836 kB
MemAvailable:    1723044 kB
Buffers:            2132 kB
Cached:          1131840 kB
SwapCached:            0 kB
Active:          1017260 kB
Inactive:         524108 kB
Active(anon):     515908 kB
Inactive(anon):    67320 kB
Active(file):     501352 kB
Inactive(file):   456788 kB
Unevictable:      107208 kB
Mlocked:               0 kB
SwapTotal:       3932156 kB
SwapFree:        3932156 kB
Dirty:                60 kB
Writeback:             0 kB
AnonPages:        495496 kB
Mapped:           298900 kB
Shmem:            175824 kB
KReclaimable:     136864 kB
Slab:             201444 kB
SReclaimable:     136864 kB
SUnreclaim:        64580 kB
KernelStack:        9268 kB
PageTables:        38896 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     5441088 kB
Committed_AS:    4513888 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       20076 kB
VmallocChunk:          0 kB
Percpu:             1184 kB
HardwareCorrupted:     0 kB
AnonHugePages:    233472 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
FileHugePages:         0 kB
FilePmdMapped:         0 kB
HugePages_Total:    2450
HugePages_Free:     2450
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:         5017600 kB
[... 241 lines cut ...]
05:00.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt 3 Bridge [Alpine Ridge 4C 2015]
05:01.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt 3 Bridge [Alpine Ridge 4C 2015]
05:02.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt 3 Bridge [Alpine Ridge 4C 2015]
05:04.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt 3 Bridge [Alpine Ridge 4C 2015]
08:00.0 USB controller: Intel Corporation DSL6540 USB 3.1 Controller [Alpine Ridge]
0a:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
0b:00.0 SATA controller: ASMedia Technology Inc. ASM1062 Serial ATA Controller (rev 02)
0e:00.0 Non-Volatile memory controller: Silicon Motion, Inc. Device 2262 (rev 03)
date
Tue Mar 10 10:30:21 CST 2020
pwd
/home/sean/pynvme
git status -sb
## master
 M .vscode/settings.json
 M driver_test.py
 M scripts/test_examples.py
 M spdk
?? api.md
git --no-pager log -1
commit 116e45e851747b80a4762fa863717c8604b83ad4
Author: Crane Chu <cranechu@gmail.com>
Date:   Thu Feb 27 15:14:12 2020 +0800

    scripts: get smart data with human readable output

cranechu, 2020-03-10:
Hi, seanmaster, 

So, you are using centos 7. Subsystem.power_cycle does not work with CentOS 7 due to the relative old kernel. Please try Fedora 31 if you can. 

If power_cycle function is not a must to have for your testing, you can still use CentOS 7, but avoid to use Subsystem.power_cycle in your scripts. 

Thanks, 
-Crane

seanmaster, 2020-03-10:
Hi cranechu,

# uname -r
5.4.1-1.el7.elrepo.x86_64

I search Fedora with kernel 5.3, and my platform is Centos7 with kernel 5.4, 
Could you have any information about kernel version support list for pynvme?
Thanks~

cranechu, 2020-03-11:
CentOS uses kernel 3.10 by default. So, you manually upgraded the kernel? 

pynvme is an userspace application, so it supports most of the kernel. However, Subsystem.power_cycle() is an exception. This function utilize system S3 power mode, as well as RTC, to power off and on NVMe devices. You can check if your system support S3 as below:

``` shell
> sudo cat /sys/power/state
freeze mem disk
> sudo cat /sys/power/mem_sleep
s2idle [deep]      ##### if you can find deep here, means the system supports S3! 
```

seanmaster, 2020-03-11:
Yes, I upgrade my os kernel to 5.x series,
my platform support S3, and I issue S3 command and resume everything is ok
"sudo rtcwake -m mem -s 10 1>/dev/null 2>/dev/null"

I will use fedora 31 and try again, after test, I will update my status. thanks~

seanmaster, 2020-03-12:
Hi cranechu,
After test on fedora31, result still fail too.
I deep trace spdk code, It is fail on nvme_ctrlr_process_init() then timeout in state 8 "NVME_CTRLR_STATE_WAIT_FOR_IDENTIFY", maybe is a compatibility issue for my test sample.
Any suggestions? very thanks~~

cranechu, 2020-03-12:
The timeout in state 8 happens after power_cycle()? If so, it means your drive did not respond the identify command after power cycle. Could you please try some different vendor's SSD? 

Thanks, 
-Crane

seanmaster, 2020-03-13:
Yes, the fail state is 8 after power_cycle(), I have 4~7 sample only 2 sample can pass the script,
I test below command in inbox driver, after S3, device response identify command Immediately,
"sudo rtcwake -m mem -s 10 1>/dev/null 2>/dev/null && nvme id-ctrl /dev/nvme0n1"

So I think it is about compatibility issue, I guess. Or could I issue relink after powercycle if first link is fail, by pyname?

cranechu, 2020-03-13:
Hi, seanmaster, 

Glad to hear that you have 2 samples can work. As a test dedicated tool, I will not focus on compatibility improvement, because it may hide some real issue of the DUT. That is the difference between pynvme and nvme-cli. So, I suggest you do some deeper analysis on those failed drives. 

You can also provide me log files of failed tests, so I can also do a further investigation. 

Thanks, 
-Crane

seanmaster, 2020-03-13:
I don't know how to attach file, so I paste full log content to here.  Below is my failed sample report. I will keep test another function with my good sample, very thank you.

make[1]: Entering directory '/home/sean/pynvme'
sudo ./spdk/scripts/setup.sh status
Hugepages
node     hugesize     free /  total
node0   1048576kB        0 /      0
node0      2048kB     2194 /   2450

NVMe devices
BDF		Vendor	Device	NUMA	Driver		Device name
0000:0e:00.0	126f	2262	0	uio_pci_generic		-

I/OAT DMA
BDF		Vendor	Device	NUMA	Driver

virtio
BDF		Vendor	Device	NUMA	Driver		Device name
VMD
BDF		Numa Node	Driver Name
sudo cat /proc/meminfo
MemTotal:        8032408 kB
MemFree:         1404128 kB
MemAvailable:    1962532 kB
Buffers:           39356 kB
Cached:           869888 kB
SwapCached:            0 kB
Active:           772352 kB
Inactive:         419004 kB
Active(anon):     420760 kB
Inactive(anon):    27144 kB
Active(file):     351592 kB
Inactive(file):   391860 kB
Unevictable:      134688 kB
Mlocked:               0 kB
SwapTotal:       8167420 kB
SwapFree:        8167420 kB
Dirty:               400 kB
Writeback:             0 kB
AnonPages:        417280 kB
Mapped:           278268 kB
Shmem:            165148 kB
KReclaimable:      72372 kB
Slab:             181552 kB
SReclaimable:      72372 kB
SUnreclaim:       109180 kB
KernelStack:        8616 kB
PageTables:        11700 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     9674824 kB
Committed_AS:    3076268 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       20684 kB
VmallocChunk:          0 kB
Percpu:             3584 kB
HardwareCorrupted:     0 kB
AnonHugePages:         0 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
CmaTotal:              0 kB
CmaFree:               0 kB
HugePages_Total:    2450
HugePages_Free:     2194
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:         5017600 kB
DirectMap4k:      190996 kB
[... 316 lines cut ...]
    ???
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

>   ???
E   nvme.NvmeEnumerateError: fail to create the controller

driver_wrap.pyx:670: NvmeEnumerateError
------------------------------ Captured log setup ------------------------------
INFO     root:conftest.py:55 running tests on DUT 0000:0e:00.0
INFO     root:conftest.py:75 bdf : b'0000:0e:00.0'
INFO     root:conftest.py:75 port : 0
INFO     root:conftest.py:68 setup random seed: 0x1041861f
------------------------------ Captured log call -------------------------------
INFO     root:test_sean.py:16 power off nvme device for 15 seconds
INFO     root:test_sean.py:16 power is back
INFO     root:test_sean.py:16 bdf : b'0000:0e:00.0'
INFO     root:test_sean.py:16 port : 0
---------------------------- Captured log teardown -----------------------------
INFO     root:conftest.py:70 test duration: 47.357 sec
=========================== short test summary info ============================
FAILED test_sean.py::test_power_cycle - nvme.NvmeEnumerateError: fail to crea...
!!!!!!!!!!!!!!!!!!!!!!!!!! stopping after 1 failures !!!!!!!!!!!!!!!!!!!!!!!!!!!
============================== 1 failed in 47.38s ==============================
make[1]: *** [Makefile:99: pytest] Error 1
make[1]: Leaving directory '/home/sean/pynvme'

cranechu, 2020-03-14:
Hi, seanmaster, 

Regarding to your log attached, pynvme works fine. It should be a device issue. 
You can also test those failed drives with another computer (desktop PC suggested). 

Thanks, 
-Crane

seanmaster, 2020-03-17:
[image]

OK , I just test another script at those failed sample, short script is above, I found the smart buffer sometimes all zero, it look like abnormal, so I add some delay after "nvme0.waitdone()" it is work, looks like those failed sample, response slowly, Any idea?

cranechu, 2020-03-17:
Yes. It seems that data transfer completes after CQ is posted. 
What's the controller of those failed drives? 

BTW, I also found the similar issue in Virtual Box about 2 years ago, and then they fixed it.  :)

seanmaster, 2020-03-23:
My sample controller is silicxxxx, maybe I should test it at another platform. now I can close the issue, very thanks~

closed 2020-03-23 by seanmaster