pynvme/pynvme #50: OSError: [Errno 24] Too manay open files issue

catogts, 2020-03-11:
Hi Crane

I use pynvme on kvm using passthrough 
(It seems to be different depending on the type of server system, but it seems to work well on the host Ubuntu 18.04 LTS, Guest Cento7 combination.but have issue msix_isset not work, i will open this issue later)

Currently, all power cycle tests are performed before ioworker close, and close after power on.  But when I test this many times, python gives too many file open error.  Is there any way to handle a lot of open files during testing?

thank you

cranechu, 2020-03-12:
Could you please give me some test log file to do further analysis? 

Thanks, 
-Crane

catogts, 2020-03-31:
Hi crane

I tested more time. But can not reproduce this

closed 2020-03-31 by catogts