pynvme/pynvme #53: Some questions  about v1.8 candidate 1

catogts, 2020-04-25:
Hi crane

I test ver 1.8 Candidate 1

I found add delay 2s to define _drive_cleanup

This delay effect to sudden power loss  test. 

Delay will give ssd time to process the accumulated data before powering down.

I see this case (fail at v1.7 , pass at v1.8 , fail at v1.8 delete delay)

And sometimes, after the ioworker terminates, the test does not proceed and stops.  

It is not known exactly what the situation is, but it is a symptom not seen in 1.7.

Be careful not to get corona virus infection

Thank you

cranechu, 2020-04-26:
Hi, Catogts, 

https://github.com/pynvme/pynvme/blob/32455e8ed5b60182700686692b37bb7ad20721f1/src/driver_wrap.pyx#L371-L389

For the 1st issue on _driver_cleanup. It does happen after power off, so its delay should not impact the power off quality. Please verify the signal of your power module if the power goes down quickly? The delay in _driver_cleanup is for main process to wait subprocesses cleaning up memory. 
BTW, we tested the sudden power loss with quarch hardware, and we can find that accumulated data loss due to the power. 

For the second problem, could you please open another issue with more information, like scripts and logs? 

Also take care and keep healthy and safe. 

Thanks, 
-Crane

catogts, 2020-05-09:
Hi crane

sorry too late confirm

first thing is my mistake. i crated poweron,off function use labjack

but cleanupdrive function place before power off. fix this no problem

second thing is not clearly reproduce process.

if i find reproduce process then open issue

thank you

cranechu, 2020-05-09:
Thanks for your feedback. 

We created a discussion group to share the knowledge of pynvme. If you use wechat and Chinese, you can join in the group by scanning the barcode below:
[image]

closed 2020-05-12 by cranechu