pynvme/pynvme #54: issue when using in a virtual machine(qemu,vfio passthrough)

catogts, 2020-05-22:
Hi crane

I am testing how to use pynvme with passthrough right now.

IOMMU+VFIO-PCI Passthrough is working correctly 

But there are things that seem to be memory related issues.

'@pytest.mark.parametrize("bs", [1,2,4,8,16,32,64,128,256])
@pytest.mark.parametrize("rw_ratio", [0,10,20,30,40,50,60,70,80,90,100])
def test_variable_bs_ratio(nvme0, nvme0n1,bs,rw_ratio):
    ioworks=nvme0n1.ioworker(io_size=bs, read_percentage=rw_ratio, time=600, lba_random=True,qdepth=1023).start().close()
    logging.info("%d, %d"%(ioworks.io_count_read, ioworks.io_count_write)
    logging.info("test complete")'

this script working good at real pc environment. but not in qemu guest

segmentation fault at test progress about 10~30%

the point at which a fault occurs is slightly different for each guest os, but it all happens.

i tested guest os centos 7, fedora 31/32 , ubuntu 19.10 

are there any suspected problems in the virtual environment?

thank you

cranechu, 2020-05-23:
Hi, Catogts, 

Could you please provide more debug information, such as:
- test.log
- terminal output around that segmentation fault
- coredump files

If you can give me a way to login to your host or even the virtual environment, we can together do a deeper debug on this issue. 

Thanks, 
-Crane

catogts, 2020-05-23:
how can i get coredump files?

cranechu, 2020-05-23:
In Fedora, you can use coredumpctl.

catogts, 2020-05-23:
ok thank you

i will try that

cranechu, 2020-05-30:
Hi, Catogts, 

Now, pynvme can test multiple devices in host OS directly, without VMs. But the code is not publicly released, and under internal testing. If you need to have an early try, I can send the prebuilt nvme.so to you. 

Thanks, 
-Crane

catogts, 2020-05-30:
Hi crane

Thanks for your suggestion.  

I want to test the functionality.  

Please file and multi-device test example.  

Also, segment fault case not occur always in vm. Still try find reproduce in absolute condition

cranechu, 2020-05-30:
OK. Please share me with your email. I will send them to you when all tests are completed. 
You can send email to: cranechu@gmail.com

catogts, 2020-05-30:
Ok 

My email is catogts@gmail.com

Thank you

catogts, 2020-06-03:
Hi crange

i found segment fault factor from virtual machine

i modified shutdown_notify  in driver_wrap.pyx

not wait rtd3e / 1000_000 (about 10sec) for measuring shutdown time

then "while (self._nvme[0x1c] & 0xc) != 0x8:" line too many access nvme register

no problem in real pc but not in vm

thank you

cranechu, 2020-06-05:
from NVMe spec:
> It is recommended that the host wait a minimum of the RTD3 Entry Latency reported in the Identify
Controller data structure for the shutdown operations to complete; if the value reported in RTD3 Entry
Latency is 0h, then the host should wait for a minimum of one second.

Have you tried to enable wait rtd3e / 1000_000 and test in vm?

Thanks, 
-Crane

catogts, 2020-06-05:
Yes 

I used it for a while when I first used pynvme in real machine

Have not use in vm

closed 2020-06-22 by cranechu