pynvme/pynvme #56: ioworker issue about sq tail over sq header

catogts, 2020-06-20:
hi crane

i tested two ioworker with reset

> nvme0n1=d.Namespace(nvme0,1,1024*1024*1024)
nvme0n1.verify_enable()
for loop in range(1000):
a=nvme0n1.ioworker(io_size=1,read_percentage=0,lba_random=True,time=20,qdepth=1023).start()
b=nvme0n1.ioworker(io_size=1,read_percentage=100,lba_random=True,time=20,qdepth=1023).start()
time.sleep(5)
nvme0.reset()
a.close()
b.close()

sorry i can not use github editor correctly 

test is two qd1023 ioworker run and reset

but pynvme failed with sq tail over sq header message

this error occur aer

thank you

catogts, 2020-06-20:
[image]
maybe not related nvme device

cranechu, 2020-06-20:
Hi, Catogts, 

I have reproduced the same symptom, and have fixed it in the latest commits. Please try your test again with the latest code. 

It is due to the asynchronous resource release between ioworker and main process.

catogts, 2020-06-22:
Hi crane

Issue cleared

Thank you!

closed 2020-06-22 by cranechu