pynvme/pynvme #63: How many ioworker can run at the same time iowoker ?

catogts, 2020-07-17:
Hi crane

I test pynvme in arm system now

Complile and running , works good!

But can not run over 32 iowoker

Is there a limit to iowoker that can be run at the same time?

Thank you

cranechu, 2020-07-17:
Hi, Catogts, 

You guys are crazy! 
As far as on x86 platform, ioworker count is only limited by devices sq/cq count and memory size. 
Maybe you can upload the load for a further analysis. 

Thanks, 
-Crane

catogts, 2020-07-17:
No need code modification :)

But ARM's SMT cpu power seems to be insufficient to run ioworker

cranechu, 2020-07-17:
> No need code modification :)
> 
> But ARM's SMT cpu power seems to be insufficient to run ioworker

It's amazing! 
For power issue, could you limit ARM's cpu core count, or limit cpu frequency? 

Thanks, 
-Crane

catogts, 2020-07-17:
> > No need code modification :)
> > 
> > But ARM's SMT cpu power seems to be insufficient to run ioworker
> 
> It's amazing! 
> Can you disable SMT and try again? 
> 
> Thanks, 
> -Crane
> 

Not yet

However, if you give a delay in the middle of creating the ioworker, it seems to be stable to some extent.

I will turn off smt next week and test it

I use Marvell Thunder X2 Platform

catogts, 2020-07-21:
SMT dis/enable not different

How much need memory for one iowoker process?

64ea iowoker make many fbarray files in /var/run/dpdk/spdkxxxxx

Failed with not found some fbarray file and "EAL : memalloc not reaponse"

Even on x86, the frequency is low, but it happens when there are many iowokers.

cranechu, 2020-07-21:
Hi, Catogts, 

One ioworker need some host buffer for data transfer. The size of this data buffer depends on MDTS and queue depth. So, you can try to reduce the qdepth of ioworkers. 

On the other hand, please use parameter *memsize* of make setup to reserve more hugepage memory: 
```shell
make setup memsize=8000  # reserve 8GB hugepage
```

closed 2020-08-08 by cranechu