pynvme/pynvme #64: Question about iowoker performance at qdepth 2

catogts, 2020-07-23:
Hi Crane.

Question about iowoker performance

i see iowoker performance data in presentation file

presendation show 4k qd=1,2 performance higher than fio

and i test 512b,4k qd=2 performance at iowoker, fio

but low qdepth performance lower than fio

What parts can you check?

i use fedora 32 and default kernel

cranechu, 2020-07-23:
Hi, can you share me your ioworker scripts and fio command lines?

catogts, 2020-07-23:
hi crane

sorry. I tested without considering the cq/sq pair (8core 16thread cpu make 16ea cq/sq pair) that the nvme module makes

fio set

[global]
bs=512
ioengine=libaio
iodepth=2
direct=1
runtime=10
filename=/dev/nvme1n1

[rand-read]
rw=randread
stonewall

fio process use 16ea cq/sq pair then performance up

i limited cq/sq pair 1ea then performance lower than pynvme

Wrong configuration of environment before testing.

closed 2020-07-29 by cranechu