pynvme/pynvme #65: Admin queue pending failed after 39d54bf

catogts, 2020-07-27:
Hi crane

Applied patch v1.9.1 (commit 39d54bf)

But iowoker failed with admin queue pending , process not found

Single iowoker no problem

However, if reset or power off occurs while running more than 10 iowokers, the above problem occurs.

No problem before

Thank you

cranechu, 2020-07-27:
Hi, Catogts, 

Thanks for the reporting. 
Could you please share us the script running to problem? We did make some changes in iowroker in this minor release. 

Thanks, 
-Crane

catogts, 2020-07-27:
no special script

io=[]

for qpair in range(63):
io.append(nvme0n1.ioworker(io_size=1,lba_random=True,read_percentage=0,qdepth=1023,time=60).start())

time.sleep(55)

subsystem.poweroff()
time.sleep(1)
subsystem.poweron()
nvme0.reset()

error message is

"ERROR nvme_pcie.c(1178) nvme_pcie_qpair_insert_pending_admin_request: The owning process (pid xxxxx) is not found. Dropping the request

maybe some iowoker not cleared

cranechu, 2020-07-27:
In 1.9.1, main process starts only when all ioworkers start to send IO. So, after time.sleep(55), some iowrokers may finished their job if the duration is 60. Please try to enlarge this durance, like 1000 sec in the example below: 

```python
def test_issue65(nvme0, nvme0n1, subsystem):
    io = []
    for qpair in range(16):
        io.append(nvme0n1.ioworker(io_size=1, lba_random=True,
                                   read_percentage=0,
                                   qdepth=1023,
                                   time=1000).start())

    time.sleep(55)
    subsystem.poweroff()
    for i in io:
        i.close()
        
    time.sleep(1)
    subsystem.poweron()
    nvme0.reset()
```

Thanks, 
-Crae

catogts, 2020-07-27:
thank you crane

i will try this tomorrow

catogts, 2020-07-29:
Thank you

Issue cleared

closed 2020-07-29 by cranechu