pynvme/pynvme #72: Need expand size to pynvme vscode plugin

catogts, 2020-09-04:
Hi crane

i have test pcie gen4 device

gen4 device bandwith over 4GB/s

but vscode plug in's performance gauge max number is 3600MB/s

So, the needle turns once and the number comes out as low as 4000

It looks like we need to change the maximum.

And are there any plans to support multi-devices in plugin? Will be very useful

Thank you

cranechu, 2020-09-04:
Hi, Catogts, Thanks for reporting this issue.

cranechu, 2020-09-06:
please try the latest code, and update the extension by: 
```shell
code --install-extension pynvme-console-2.0.0.vsix --force
```

cranechu, 2020-09-06:
For multiple device support in vscode, we don't have any plan for now.
VSCode is an environment for debugging scripts, and we suggest to run tests in command line environment. 

Thanks!

catogts, 2020-09-07:
Thank you crane

I tested fixed module

But not show real speed

Iops is correct. But bandwidth is lower about 4000

It doesn’t rise after 4000, it seems to go up again from 0.

cranechu, 2020-09-07:
> Thank you crane
> 
> I tested fixed module
> 
> But not show real speed
> 
> Iops is correct. But bandwidth is lower about 4000
> 
> It doesn’t rise after 4000, it seems to go up again from 0.

Thanks! It is caused by data type of uint32_t, so cap at 4GB/s. Fixed it. 
Please pull pynvme, compile and try it again.

catogts, 2020-09-08:
Works good!

Thank you

closed 2020-09-08 by cranechu