pynvme/pynvme #74: Hello World Test failed

sprotznock85, 2020-09-24:
Hi,

i installed pynvme via the install.sh script on Ubuntu but I can't get the first test working. Following error occurs:

```
sudo python3 -B -m pytest scripts/test_examples.py::test_hello_world --pciaddr=0000:0a:00.0 -s -v -r Efsx --excelreport=report.xls --verbose -Werror

=================================================================== test session starts ====================================================================
platform linux -- Python 3.7.9, pytest-6.0.2, py-1.9.0, pluggy-0.13.1 -- /usr/bin/python3
cachedir: .pytest_cache
rootdir: /home/labor/hendrik.freund/pynvme, configfile: pytest.ini
plugins: excel-1.4.1, cov-2.10.1
collected 1 item

scripts/test_examples.py::test_hello_world
---------------------------------------------------------------------- live log setup ----------------------------------------------------------------------
[2020-09-24 14:52:49.105] INFO script(60): setup random seed: 0xad6bc654
EAL: Driver cannot attach the device (0000:0a:00.0)
EAL: Failed to attach device on primary process
[2020-09-24 14:52:49.107258] ERROR nvme.c(567) spdk_nvme_probe_internal: NVMe ctrlr scan failed
[2020-09-24 14:52:49.107272] ERROR nvme.c(640) spdk_nvme_probe: Create probe context failed
[2020-09-24 14:52:49.107277] WARNING driver.c(971) nvme_probe: not found device: 0000:0a:00.0, rc -1, cb_ctx.ctrlr (nil)
ERROR
-------------------------------------------------------------------- live log teardown ---------------------------------------------------------------------
[2020-09-24 14:52:49.125] INFO script(62): test duration: 0.020 sec

========================================================================== ERRORS ==========================================================================
____________________________________________________________ ERROR at setup of test_hello_world ____________________________________________________________

pciaddr = '0000:0a:00.0'

    @pytest.fixture(scope="function")
    def pcie(pciaddr):
>       ret = d.Pcie(pciaddr)

conftest.py:72:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

>   ???
E   nvme.NvmeEnumerateError: fail to create the controller

driver_wrap.pyx:556: NvmeEnumerateError
-------------------------------------------------------------------- Captured log setup --------------------------------------------------------------------
INFO     root:conftest.py:60 setup random seed: 0xad6bc654
----------------------------------------------------------------- excel report: report.xls -----------------------------------------------------------------
================================================================= short test summary info ==================================================================
ERROR scripts/test_examples.py::test_hello_world - nvme.NvmeEnumerateError: fail to create the controller
===================================================================== 1 error in 0.09s =====================================================================
```

I do run the test on the NVME itself. Is this even possible or do I need an extra drive to run the tests?
BR

cranechu, 2020-09-24:
could you please run the commandline
```shell
make setup
```
and then paste the output log?

Thanks!

sprotznock85, 2020-09-24:
```
make setup
sudo sh -c "echo 1 > /sys/bus/pci/rescan"
sudo rm -rf /run/dpdk
sudo rm -rf /var/run/dpdk
sudo ./src/setup.sh cleanup
Clean
sudo ./src/setup.sh reset
0000:0a:00.0 (8086 f1a5): Already using the nvme driver
sudo rm -f /var/tmp/spdk.sock*
sudo rm -f /var/tmp/pynvme.sock*
sudo rm -rf .pytest_cache
sudo fuser -k 4420/tcp
Makefile:65: recipe for target 'reset' failed
make: [reset] Error 1 (ignored)
sudo sh -c 'find . | grep -E "(__pycache__|\.pyc|\.pyo$)" | xargs rm -rf'
xhost +local:           # enable GUI with root/sudo
non-network local connections being added to access control list
sudo chmod 777 /tmp
sudo sh -c 'find . | grep -E "(__pycache__|\.pyc|\.pyo$)" | xargs rm -rf'
sed -i 's/XXXX:BB:DD.F/0000:0a:00.0/g' .vscode/settings.json
sudo HUGEMEM=2430    DRIVER_OVERRIDE=uio_pci_generic ./src/setup.sh     # UIO is recommended
0000:0a:00.0 (8086 f1a5): Active mountpoints on /dev/nvme0n1, so not binding PCI dev
```

cranechu, 2020-09-24:
```shell
0000:0a:00.0 (8086 f1a5): Active mountpoints on /dev/nvme0n1, so not binding PCI dev
```

Your only NVMe drive has active partitions. Is it your OS drive? We cannot run tests on that because it may overwrite your data. You need another NVMe disk for testing only.

sprotznock85, 2020-09-25:
Hi,

Yes it is the OS system. Thanks for that.
BR

closed 2020-09-25 by cranechu