pynvme/pynvme #77: suggest function for zns test

catogts, 2020-10-12:
hi crane

i am trying the zns test function

The test works normally.  However, since qpair is connected to the zone management command, there is a problem that script have to delete the qpair in advance when performing a reset or power cycle.

Like Python space driver or ioworker, I think need an option to ignore qpair when reset or power cycle.

thank you

cranechu, 2020-10-13:
Hi, Catogts, 

Thanks for your feedback. Yes, it is an important notice to delete all qpairs before performing reset or power cycle. Currently, we have to do it in user scripts. We will try to find a better way in future releases. 

BTW, did you test zns on emulated QEMU devices or physical SSD? 

Thanks, 
-Crane

catogts, 2020-10-14:
I use qemu emulate drive
I know that some physical zns drive  use customized spec

yfwayne, 2020-10-14:
Hi, catogts, I updated scripts such that it could work on emulated QEMU ZNS drive first. Since you are the among the first few user of the ZNS package, I would warmly invite you to join and contribute to it and make it better if possible. As you can see, it is still in its early stage.

cranechu, 2020-10-15:
Hi, Catogts, 

Yuanfang is the owner of zns test scripts (pynvme/scripts/conformance/07_zns). We are testing ZNS scripts with emulated devices (https://gist.github.com/birkelund/22f36aa4c4411525f50478c4f9c192ab). Please confirm if we are using the same emulated ZNS? 

Thanks, 
-Crane

catogts, 2020-10-15:
Hi crane, Hi Yuanfang

I got the qemu zns driver through patchworks.  I don't remember who the creator is.  I'll check it tomorrow.

catogts, 2020-10-19:
Ok~ I confirmed that I use the same qemu zns driver

cranechu, 2020-12-05:
All qpair created by scripts have to be deleted before performing reset or powercycle.

closed 2020-12-05 by cranechu