pynvme/pynvme #78: can not issue io cmd at special case

catogts, 2020-10-14:
Hi crane

i found some special case

can not issue io cmd to nvme

code is below

`qpairs={}
 qpairs_count={}
 iops_count=0
 buf=d.Buffer(512*8)
 for a in range(64):
        qpairs[a]=d.Qpair(nvme0,64+1)
        qpairs_count[a]=0
 for a in range(100000):
        choice=random.randrange(len(qpairs))
        lba=random.randrange(io_count)
        if qpairs_count[choice] < 64:
             nvme0n1.write(qpairs[choice],buf,lba,8)
             qpairs_count[choice]+=1
             iops_count+=1
        if iops_count > 1000:
             for key,value in qpairs.items():
                    value.waitdone(qpairs_count[key])
                    qpairs_count[key]=0
             iops_count=0`

Some of the write cmds are not registered in qpair and a timeout error occurs in qpair waitdone.

But if I change lba sequentially, there is no problem.  

Are there any restrictions on using random lba when using multiple qpairs?

thank you!

cranechu, 2020-10-15:
```python
def test_random_io(nvme0, nvme0n1):
    qpairs = {}
    qpairs_count = {}
    iops_count = 0

    import random

    buf = d.Buffer(512*8)
    for a in range(64):
        qpairs[a] = d.Qpair(nvme0, 64+1)
        qpairs_count[a] = 0

    io_count = 100000
    for a in range(io_count):
        choice = random.randrange(len(qpairs))
        lba = random.randrange(io_count)
        if qpairs_count[choice] < 64:
            nvme0n1.write(qpairs[choice], buf, lba, 8)
            qpairs_count[choice] += 1
            iops_count += 1
        if iops_count > 1000:
            for key, value in qpairs.items():
                value.waitdone(qpairs_count[key])
                qpairs_count[key] = 0
            iops_count = 0

    for k, v in qpairs.items():
        v.delete()

```

catogts, 2020-10-15:
I tested samsung and other drive

What version of the kernel and Linux have you tested?

I think I have to test it in another system environment.

Thank you

cranechu, 2020-10-15:
I am using the latest updated Fedora 32, but I don't think it makes the differences. 
Could you please upload the test log? 

Thanks, 
-Crane

catogts, 2020-10-15:
hi crane

I have found one of the causes.  

Have you set enough memory to enable data verification during the test? (make setup memsize=xxxx) 

If you test the entire area without enough memory to verify data verification, the test will run without problems.  

But if you set enough memory, problems arise.  

The error log is simply the pynvme driver time out.  

Occurs because waitdone(x) is waiting more than the actual stacked queue cmd.

This problem is not affected by the nvme0n1.verify_enable option and I also checked it on the samsung 960 evo

thank you

cranechu, 2020-10-16:
Understand. I can also reproduce the issue. It is caused by the LBA lock. 

Because NVMe IO are handled out of order, data verification does not know what is the latest data. We introduced the LBA lock to handle this problem. The lock is applied when sending commands, and released in waitdone when the command is completed. 

In your script, for example, we send 100 write commands to a qpair, but some of them may be blocked due to the LBA conflict. So, the actual command count may be less than 100. That is the reason we got the waitdone timeout. 

My suggestion: we should clear LBA conflict in our script. One way is to use sequential IO you mentioned in the first message. Or, we can make the script a little bit more complex: we re-generate random LBA address when that address is the same as earlier addressed generated. 

Thanks, 
-Crane

catogts, 2020-10-19:
Thank you crane!

I confirmed lba conflict issue

I think that testing by reducing the verify area rather than checking duplicate lba will help speed, so I will use it.

cranechu, 2020-10-19:
> Thank you crane!
> 
> I confirmed lba conflict issue
> 
> I think that testing by reducing the verify area rather than checking duplicate lba will help speed, so I will use it.

good idea!

closed 2020-10-23 by cranechu