pynvme/pynvme #81: Is there any limit on the number of lba?

catogts, 2020-11-13:
Hi crane

During the 8TB nvme test, i checked a problem with verification.  

target ssd lba size is 512 bytes, the number of lba is about 14 billion. 

If i load the namespace, eal_memalloc_alloc_seg_bulk error occurs even though there is enough hugepage.  

Loading 4 4TB ssds does not cause the problem, but loading only 1 8TB ssd causes the above problem

Is there a limit to the number of lba?

thank you

cranechu, 2020-11-20:
Please try this: 

> The administrator can allocate persistent huge pages on the kernel boot
command line by specifying the "hugepages=N" parameter, where 'N' = the
number of huge pages requested.  This is the most reliable method of
allocating huge pages as memory has not yet become fragmented.

from: https://www.kernel.org/doc/Documentation/vm/hugetlbpage.txt

catogts, 2020-11-26:
The hugepage doesn't seem to be relevant. It is assumed that SPDK is not able to allocate memory for a size larger than a certain size (probably 64GB or more).

closed 2021-01-06 by cranechu