pynvme/pynvme #90: error when pip install

yunyuyuan, 2020-12-07:
```
ERROR: Command errored out with exit status 1:
     command: 'C:\Users\yunyuyuan\Desktop\pyhwmonitor\venv\Scripts\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\YUNYUY~1\\AppData
\\Local\\Temp\\pip-install-izxfk0t2\\pynvme\\setup.py'"'"'; __file__='"'"'C:\\Users\\YUNYUY~1\\AppData\\Local\\Temp\\pip-install-izxfk0t2\\pynvme\\setup.py'"'"';f=g
etattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_in
fo --egg-base pip-egg-info
         cwd: C:\Users\YUNYUY~1\AppData\Local\Temp\pip-install-izxfk0t2\pynvme\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\YUNYUY~1\AppData\Local\Temp\pip-install-izxfk0t2\pynvme\setup.py", line 47, in <module>
        long_description = fh.read()
    UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 1812: illegal multibyte sequence
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

cranechu, 2020-12-07:
Hi, 

Please follow this webpage to install pynvme in Linux. 
https://pynvme.readthedocs.io/install.html

Thanks, 
-Crane

yunyuyuan, 2020-12-08:
> Hi,
> 
> Please follow this webpage to install pynvme in Linux.
> https://pynvme.readthedocs.io/install.html
> 
> Thanks,
> -Crane

Okey,It seems that pynvme can only run on Linux.

closed 2020-12-08 by yunyuyuan