pynvme/pynvme #93: pynvme fails when running in standalone mode

xsummusx, 2020-12-29:
Downloaded and installed fresh build of pynvme today following instructions:
```
git clone https://github.com/pynvme/pynvme
cd pynvme
./install.sh
```
I then attempted running the following commands and received these errors:
```
Python 3.8.5 (default, Sep  4 2020, 07:30:14) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import nvme as d
>>> d.Pcie('3c:00.0')
EAL: memzone_reserve_aligned_thread_unsafe(): No more room in config
[2020-12-28 20:16:13.100020] ERROR nvme.c(347) nvme_driver_init: primary process failed to reserve memory
[2020-12-28 20:16:13.100063] ERROR nvme.c(640) spdk_nvme_probe: Create probe context failed
[2020-12-28 20:16:13.100103] WARNING driver.c(1001) nvme_probe: not found device: 0000:3c:00.0, rc -1, cb_ctx.ctrlr (nil)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "driver_wrap.pyx", line 562, in nvme.Pcie.__cinit__
nvme.NvmeEnumerateError: fail to create the controller
>>> d.Buffer(1024, 'example')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "driver_wrap.pyx", line 252, in nvme.Buffer.__cinit__
MemoryError
>>> 
```
I am running Ubuntu:
```
Linux xxxxxxxxx-NUC7i5BNK 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
I have tried increasing the **CONFIG_RTE_MAX_MEMZONE=5120** value in dpdk and rebuilt but got the same errors based on searches to the error **EAL: memzone_reserve_aligned_thread_unsafe(): No more room in config**.  

It is quite possible I don't know what I am doing here but from what I can tell from the documentation this should work.  BTW, lspci give the following with my DUT being 3c:00.0:
```
00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 03)
00:02.0 VGA compatible controller: Intel Corporation Iris Plus Graphics 640 (rev 06)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.5 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 (rev f1)
00:1c.7 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #8 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Sunrise Point LPC Controller/eSPI Controller (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (4) I219-V (rev 21)
3a:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)
3b:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5229 PCI Express Card Reader (rev 01)
3c:00.0 Non-Volatile memory controller: Micron Technology Inc Device 5410 (rev 01)
```
I checked that the drive is functional using fio and reverting the uio driver back to the nvme driver.  Don't know if it matters but I am using Anaconda for the python environment.
```
(base) xxxxxxxx@xxxxxxxx-NUC7i5BNK:~/src/pynvme/spdk/scripts$ sudo ./setup.sh reset
0000:3c:00.0 (1344 5410): uio_pci_generic -> nvme
(base) xxxxxxxx@xxxxxxxx-NUC7i5BNK:~/src/pynvme/spdk/scripts$ sudo nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1     18361E4E2DE8         Micron_2200_EEFDHBA1T0TCK                1           1.02  TB /   1.02  TB    512   B +  0 B   P1FB005 
```
Thoughts on what I am doing wrong here?

cranechu, 2020-12-29:
I did not test pynvme with Anaconda, but I don't think it is a problem. Please make sure you started python3 with **sudo**

Could you please upload the log of ./install.sh? 
And then, please try commands below, and also upload the generated test.log:

``` python
make setup
make test TESTS="driver_test.py::test_ioworker_iops_multiple_queue[1]"
```

Thanks!

xsummusx, 2020-12-29:
I captured the output from the steps you recommended above and put the output below.  As you can see below, importing with sudo python3 did not work and failed to load DPDK/SPDK.  However, I tried it again, sudo python3, and now it is working.  I suspect it may have to do with **~/src/pynvme/spdk/scripts$ sudo ./setup.sh reset** I ran earlier or one of the commands you gave me to run.  I will mess with it more tomorrow and report back.

I started python3 using sudo with the following results:
```
(base) xxxxxxxx@xxxxxxxx-NUC7i5BNK:~/temp/pynvme$ sudo python3
[sudo] password for xxxxxxxx: 
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import nvme as d
EAL: No free hugepages reported in hugepages-2048kB
EAL: FATAL: Cannot get hugepage information.
EAL: Cannot get hugepage information.
Failed to initialize DPDK
Unable to initialize SPDK env
ERROR:root:driver initialization fail
driver initialization fail
```

I should have mentioned in my previous post the BIOS configuration:
```
RAID mode (Intel® RST): shall be disabled in BIOS.  **<- disabled**
Secure boot: shall be disabled in BIOS.  **<- disabled**
IOMMU: (a.k.a. VT for Direct I/O) shall be disabled in BIOS.  **<- disabled**
VMD: shall be disabled in BIOS.  **<- not an option provided by the BIOS**
NUMA: recommend to be disabled in BIOS.  **<- not an option provided by the BIOS**
```
**install.sh log:**
[test_0000:3c:00.0.log](https://github.com/pynvme/pynvme/files/5750017/test_0000.3c.00.0.log)

make test TESTS="driver_test.py::test_ioworker_iops_multiple_queue[1]" results:
<it won't let me upload this file for some reason, so here are the last lines>

```
git status -sb
## master...origin/master
 M .vscode/settings.json
?? report.xls
?? test.txt
git --no-pager log -1
commit c082e80974327ddb1a69fbf4a3146c1ffb2cb496
Author: Crane Chu <cranechu@gmail.com>
Date:   Wed Dec 16 16:25:03 2020 +0800

    update pynvme3 features and contacts
sudo python3 -B -m pytest driver_test.py::test_ioworker_iops_multiple_queue[1] --pciaddr=0000:3c:00.0 -s -v -r Efsx --excelreport=report.xls --verbose
============================= test session starts ==============================
platform linux -- Python 3.8.5, pytest-6.0.1, py-1.9.0, pluggy-0.13.1 -- /usr/bin/python3
cachedir: .pytest_cache
rootdir: /home/xxxxxxxx/temp/pynvme, configfile: pytest.ini
plugins: cov-2.10.1, excel-1.4.2
collecting ... collected 1 item

driver_test.py::test_ioworker_iops_multiple_queue[1] 
-------------------------------- live log setup --------------------------------
[2020-12-29 00:20:39.490] INFO script(60): setup random seed: 0x10dd9ff5
-------------------------------- live log call ---------------------------------
[2020-12-29 00:20:49.812] INFO test_ioworker_iops_multiple_queue(2788): {'io_count_read': 2474363, 'io_count_nonread': 0, 'mseconds': 10001, 'latency_max_us': 24156, 'error': 0, 'cpu_usage': 0.24957504249575044, 'latency_average_us': 252, 'io_count_write': 0}
[2020-12-29 00:20:49.813] INFO test_ioworker_iops_multiple_queue(2791): Q 1 IOPS: 247.436K
PASSED
------------------------------ live log teardown -------------------------------
[2020-12-29 00:20:49.815] INFO script(62): test duration: 10.325 sec

--------------------------- excel report: report.xls ---------------------------
============================== 1 passed in 10.58s ==============================
```

cranechu, 2020-12-29:
It looks good. You have setup your platform properly. Just remember to run **make setup** before tests.

xsummusx, 2020-12-29:
Thank you.  This was my mistake not running **make setup** before the test, which is the magic that switches out the nvme driver with the SPDK uio driver.  This did get done once during my install but I rebooted along the way and reset my environment.

Can I suggest some additional documentation at [https://pynvme.readthedocs.io/intro.html](url)?  I am only able to find text about running **make setup** on the Install page and all the examples are for pytest.  My suggestion is to add a couple of examples in addition to the pytest examples showing standalone pynvme development using **make setup**  and **make reset**.  I would be happy to write up the additional documentation and submit for your approval if that will help you out - just let me know.

closed 2020-12-29 by xsummusx

cranechu, 2020-12-30:
Sure, please go ahead. You can update the files in the directory doc, and
send me the PR.

Thanks!