pynvme/pynvme #96: Controller(Pcie('01:00.0')) failing

pratdandamudi, 2021-01-21:
Hello, 

I'm trying to initialize test NVMe controller but the initialization times out

Below is the error message:
------------------------------------------------------------------------------------------------
<console>:1: UserWarning: drive timeout: qpair: 0, cid: 126
<console>:1: UserWarning: ERROR status: 07/ff
Traceback (most recent call last):
  File "<console>", line 1, in <module>
  File "driver_wrap.pyx", line 906, in nvme.Controller.__cinit__
  File "driver_wrap.pyx", line 946, in nvme.Controller._nvme_init
  File "driver_wrap.pyx", line 1160, in nvme.Controller.waitdone
  File "driver_wrap.pyx", line 93, in nvme._timeout_signal_handler
TimeoutError: pynvme timeout in driver

--------------------------------------------------------------------------------------------------

The boot drive for my linux PC is also a NVMe drive. Does this have anything to do with the error?

Thanks,
Prathyusha

cranechu, 2021-01-21:
Could you please give us a full log generated by:

```shell
make setup
make test TESTS="driver_test.py::test_ioworker_iops_multiple_queue[1]"
```

pratdandamudi, 2021-01-21:
controller on 04:00.0 is the NVMe boot drive

cranechu, 2021-01-21:
So, try this command line to specify BDF in the test:

```shell
make setup
make test TESTS="driver_test.py::test_ioworker_iops_multiple_queue[1]" pciaddr=0000:01:00.0
```

pratdandamudi, 2021-01-22:
Thanks for your suggestion

I specified the BDF in the test and below is the output: Got timeout error

make test TESTS="driver_test.py::test_ioworker_iops_multiple_queue[1]" pciaddr=0000:01:00.0
rm test_0000:01:00.0.log
make pytest 2>test_0000:01:00.0.log | tee -a test_0000:01:00.0.log
make[1]: Entering directory '/home/prathyusha/pynvme'
sudo ./src/setup.sh status
Hugepages
node     hugesize     free /  total
node0   1048576kB        0 /      0
node0      2048kB     1215 /   1215

NVMe devices
BDF		Vendor	Device	NUMA	Driver		Device name
0000:01:00.0	1b4b	1325	0	uio_pci_generic		-
0000:04:00.0	144d	a808	0	nvme			nvme1

I/OAT DMA
BDF		Vendor	Device	NUMA	Driver

virtio
BDF		Vendor	Device	NUMA	Driver		Device name
VMD
BDF		Numa Node	Driver Name
sudo cat /proc/meminfo
MemTotal:        6075636 kB
MemFree:         1076976 kB
MemAvailable:    1939536 kB
Buffers:           59984 kB
Cached:           985632 kB
SwapCached:            0 kB
Active:          1274128 kB
Inactive:         782420 kB
Active(anon):    1012968 kB
Inactive(anon):    21440 kB
Active(file):     261160 kB
Inactive(file):   760980 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:      19530748 kB
SwapFree:       19530748 kB
Dirty:              1044 kB
Writeback:           148 kB
AnonPages:       1010940 kB
Mapped:           447432 kB
Shmem:             23484 kB
KReclaimable:     103192 kB
Slab:             237708 kB
SReclaimable:     103192 kB
SUnreclaim:       134516 kB
KernelStack:       13120 kB
PageTables:        17920 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:    21324404 kB
Committed_AS:    5180148 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       37592 kB
VmallocChunk:          0 kB
Percpu:            28800 kB
HardwareCorrupted:     0 kB
AnonHugePages:         0 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
FileHugePages:         0 kB
FilePmdMapped:         0 kB
HugePages_Total:    1215
[... 357 lines cut ...]
    ???
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

>   ???
E   TimeoutError: pynvme timeout in driver

driver_wrap.pyx:93: TimeoutError
------------------------------ Captured log setup ------------------------------
INFO     root:conftest.py:60 setup random seed: 0x1fef101a
=============================== warnings summary ===============================
driver_test.py::test_ioworker_iops_multiple_queue[1]
  /home/prathyusha/pynvme/conftest.py:79: UserWarning: drive timeout: qpair: 0, cid: 126
    ret = d.Controller(pcie)

driver_test.py::test_ioworker_iops_multiple_queue[1]
  /home/prathyusha/pynvme/conftest.py:79: UserWarning: ERROR status: 07/ff
    ret = d.Controller(pcie)

-- Docs: https://docs.pytest.org/en/stable/warnings.html
--------------------------- excel report: report.xls ---------------------------
=========================== short test summary info ============================
ERROR driver_test.py::test_ioworker_iops_multiple_queue[1] - TimeoutError: py...
======================== 2 warnings, 1 error in 30.32s =========================
make[1]: Leaving directory '/home/prathyusha/pynvme'
sudo rm -rf .pytest_cache

cranechu, 2021-01-25:
Can you try another drive from different vendor? 
From the log, it seems a drive issue.

closed 2021-02-03 by cranechu