pynvme/pynvme #100: The parameters of the "get_cmdlog" API?

yclin1209, 2021-04-06:
Thanks for your excellent work on pynvme, 
It is very helpful for me to develop SSD test program.

I do some modified on cmdlog, but I can't call get_cmdlog correct. 
Always get errors such as invalid parameter. 
How can I transfer its parameter correctly?
e.g.: result = jsonrpc_call(sock, 'get_cmdlog','???') 

Thank You. 
Regards

cranechu, 2021-04-06:
Thanks for using pynvme!
The jsonrpc call "get_cmdlog" has only one parameter, and has to be integer value of a valid queue id. e.g.: 
```python
qpair = ...
result = jsonrpc_call(sock, 'get_cmdlog', qpair.sqid)
```

yclin1209, 2021-04-06:
Thank you for your quick reply.
I insert the code snippet to test_examples.py:test_jsonrpc_list_qpairs(),
but the jsonrpc_call(sock, 'get_cmdlog', q1.sqid) still got error: "invalid request" .
I also try integer: 0~2, but  still got "invalid request".
Am I missing anything else necessary? 

 `q1 = d.Qpair(nvme0, 8)
    result = jsonrpc_call(sock, 'get_cmdlog', q1.sqid)`

cranechu, 2021-04-07:
please refer to this piece of script:
``` python
    q1 = d.Qpair(nvme0, 8)
    result = jsonrpc_call(sock, 'list_all_qpair')
    assert len(result) == 2
    assert result[0]['qid']-1 == 0
    assert result[1]['qid']-1 == 1
    cmdlog = jsonrpc_call(sock, 'get_cmdlog', [result[0]['qpair']])
    for c in cmdlog:
        logging.info(c)
    cmdlog = jsonrpc_call(sock, 'get_cmdlog', [result[1]['qpair']])
    assert cmdlog == []
```

Our vscode extension is also built on top of these jsonrpc calls. 

Thanks, 
-Crane

yclin1209, 2021-04-07:
> please refer to this piece of script:
>
> ```python
>     q1 = d.Qpair(nvme0, 8)
>     result = jsonrpc_call(sock, 'list_all_qpair')
>     assert len(result) == 2
>     assert result[0]['qid']-1 == 0 [...]

   It's works!
   Thanks for your help.

closed 2021-04-07 by yclin1209