pynvme/pynvme #101: HMB buffer entries mapped to same phy_addr

pratdandamudi, 2021-05-03:
Hi Crane,

I'm testing my HMB set feature implementation using pynvme. I assign total size of 0x400 in host MPS so a total of 4MB among 4 buffer entries. I noticed that the alternate buffer entries are mapped to same physical memory. Is there a way to assign separate memory to each buffer entry?

Below is my code for HMB set feature: 

def hmb_feature(en,hsize,num_entries):

    hmb_entry_count = num_entries
    hmb_buf_size_pages = math.floor(hsize/hmb_entry_count) #round down if not divisible by hmb_entry_count
    hsize = hmb_buf_size_pages*hmb_entry_count

    hmb_buf_size_bytes = hmb_buf_size_pages*4096 #4096 is MPS
    hmb_dl_entry_size = 16 #in bytes

    hmb_buf_list=[]

    hmb_dl_buf = d.Buffer(hmb_entry_count*hmb_dl_entry_size)
    hmb_dl_lower_addr = hmb_dl_buf.phys_addr&0xffffffff
    hmb_dl_upper_addr = (hmb_dl_buf.phys_addr>>32)&0xffffffff

    for i in range(hmb_entry_count):
        hmb_buf = d.Buffer(hmb_buf_size_bytes)
        hmb_buf_lower_addr = hmb_buf.phys_addr&0xffffffff
        hmb_buf_upper_addr = (hmb_buf.phys_addr>>32)&0xffffffff

        print("hmb buffer {:02X} phy_addr lower = 0x{:04X} upper = 0x{:04X}, buffer size in MPS = 0x{:04X}\n".format(i, hmb_buf_lower_addr, hmb_buf_upper_addr, hmb_buf_size_pages))

        #BADD lower 32 bits
        hmb_dl_buf[hmb_dl_entry_size*i] = (hmb_buf_lower_addr>>0)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+1] = (hmb_buf_lower_addr>>8)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+2] = (hmb_buf_lower_addr>>16)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+3] = (hmb_buf_lower_addr>>24)&0xff

        #BADD upper 32 bits
        hmb_dl_buf[(hmb_dl_entry_size*i)+4] = (hmb_buf_upper_addr>>0)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+5] = (hmb_buf_upper_addr>>8)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+6] = (hmb_buf_upper_addr>>16)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+7] = (hmb_buf_upper_addr>>24)&0xff

        #BSIZE
        hmb_dl_buf[(hmb_dl_entry_size*i)+8] = (hmb_buf_size_pages>>0)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+9] = (hmb_buf_size_pages>>8)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+10] = (hmb_buf_size_pages>>16)&0xff
        hmb_dl_buf[(hmb_dl_entry_size*i)+11] = (hmb_buf_size_pages>>24)&0xff

        #reserved bits
        hmb_dl_buf[(hmb_dl_entry_size*i)+12] = 0x0
        hmb_dl_buf[(hmb_dl_entry_size*i)+13] = 0x0
        hmb_dl_buf[(hmb_dl_entry_size*i)+14] = 0x0
        hmb_dl_buf[(hmb_dl_entry_size*i)+15] = 0x0

    #Controller.setfeatures(self, fid, sv=0, buf=None, cdw11=0, cdw12=0, cdw13=0, cdw14=0, cdw15=0, cb=None)
    #cdw11[0:0] = EHM. cdw11[1:1] = MR, cdw12 = hmb size, cdw13 = HMDLLA, cdw14 = HMDLUA, cdw15 = HMDLEC
    cdw11 = en
    cdw12 = hsize
    cdw13 = hmb_dl_lower_addr
    cdw14 = hmb_dl_upper_addr
    cdw15 = hmb_entry_count

    ctrl.setfeatures(13, 0, None, cdw11, cdw12, cdw13, cdw14, cdw15, None)
    #need to have a ctrl.waitdone() call to update the CQ head DB
    ctrl.waitdone()
    print("HMB set feature command sent! cdw11 - cdw15 0x{:04X} 0x{:04X} 0x{:04X} 0x{:04X} 0x{:04X}".format(cdw11, cdw12, cdw13, cdw14, cdw15))

Below is the log after running the code: 

>>> hmb_feature(1,0x400,0x4)
hmb buffer 00 phy_addr lower = 0x5F27A000 upper = 0x0003, buffer size in MPS = 0x0100

hmb buffer 01 phy_addr lower = 0x5F09A000 upper = 0x0003, buffer size in MPS = 0x0100

hmb buffer 02 phy_addr lower = 0x5F27A000 upper = 0x0003, buffer size in MPS = 0x0100

hmb buffer 03 phy_addr lower = 0x5F09A000 upper = 0x0003, buffer size in MPS = 0x0100

HMB set feature command sent! cdw11 - cdw15 0x0001 0x0400 0x5F37B000 0x0003 0x0004

Thanks a lot
-Prathyusha

cranechu, 2021-05-04:
Hi, Prathyusha, 

As stated in the feature list (https://github.com/pynvme/pynvme#features), HMB is only supported in PyNVMe3 (http://pynv.me/3/). 

Thanks, 
-Crane

pratdandamudi, 2021-05-04:
Thanks, I will checkout the pynvme3

closed 2021-07-26 by cranechu