pynvme/pynvme #102: About Class Controller() issue

shampoo365, 2021-05-18:
I got a problem when I use the class Controller() but I am not if it makes sense or not, since from my professional aspect as an SSD FW engineer, it seems to be an issue.

The details are below:
1. I have a SSD supporting multi-namepace, NN in Identify is 8.
2. The host system only create one namespace and attached it to the controller, reporting as /dev/nvme8n1
3. However, when I use class Controller(), it got lots of print that from SPDK, which seems to **try to identify NS from 2 to 8**.
4. So I think the identify operation should be triggered by calling Controller().
5. Can it be optimzed? I think if it only has one NS, we should only identify NS1, rather than 2 to 8

cranechu, 2021-05-18:
Thanks very much for your feedback. It's true that we have to optimized this process for enterprise applications. We have started the job in another project PyNVMe3, and pynvme will continue to be maintained for mainstream client SSD test.

shampoo365, 2021-05-19:
> Thanks very much for your feedback. It's true that we have to optimized this process for enterprise applications. We have started the job in another project PyNVMe3, and pynvme will continue to be maintained for mainstream client SSD test.

Ok, I see. I would switch to PyNVMe3. By the way, it is for enterprise or DC-SSD, right? Because my work is on enterprise/DC SSD.
Another question is, since I am not a python expert like you, if I write a python file, do I have to put it into ./script/ to make run?

shampoo365, 2021-05-19:
Is PyNVMe3 open source?

cranechu, 2021-05-19:
> Is PyNVMe3 open source?

PyNVMe3 is our commercial product with different Editions. It involves more features and services. Its driver source code is closed, while all python code files (both library and tests) are open to our customers. 
Please find more information of PyNVMe3 here: http://pynv.me/3/

Thanks!

shampoo365, 2021-05-19:
As for pynvme, is it open for driver source code? BTW, the driver source code is not SPDK nvme driver?

cranechu, 2021-05-19:
pynvme driver source code is open. pynvme driver is developed on SPDK, but it is not the original SPDK nvme driver. We did some development inside and outside of SPDK.

cranechu, 2021-05-19:
You can start with pynvme, and we can discuss your findings and questions here.

shampoo365, 2021-05-20:
> You can start with pynvme, and we can discuss your findings and questions here.

Hello, I have a question about the source code of pynvme.
I found the submodule SPDK is a very old version and from gitee two years ago when I did 'git submodule update --init', however, from the source code drive.c, it seems some functions are based on a new SPDK from github.
Does that mean the functions above cannot work?

Thanks,

cranechu, 2021-05-20:
Hi, 

Our SPDK modified from version v19.10. What functions are you talking about?

shampoo365, 2021-05-20:
> Hi,
> 
> Our SPDK modified from version v19.10. What functions are you talking about?

For example,
The function **nvme_pcie_qpair_reset()** called by **function nvme_set_adminq()** in _driver.c_ could not be found in the pynvme source code, but it could be found in _lib/nvme/nvme_pcie_common.c_ SPDK from github.
However, pynvme's spdk has no lib/nvme/nvme_pcie_common.c

closed 2021-07-26 by cranechu