pynvme/pynvme #104: memzone_reserved_aligned error when doing nvme0 = d.Controller(d.Pcie("02:00.0"))

nate12112, 2021-06-02:
I keep getting this error when I do nvme0 = d.Controller(d.Pcie("03:00.0"))

EAL: memzone_reserve_aligned_thread_unsafe(): No more room in config
[2021-06-02 15:08:03.846887] ERROR nvme.c(347) nvme_driver_init: primary process failed to reserve memory
[2021-06-02 15:08:03.846915] ERROR nvme.c(640) spdk_nvme_probe: Create probe context failed
[2021-06-02 15:08:03.846933] WARNING driver.c(1002) nvme_probe: not found device: 0000:03:00.0, rc -1, cb_ctx.ctrlr (nil)

cranechu, 2021-06-03:
Could you please share the whole log file, so we can find more system information. Thanks!

nate12112, 2021-06-03:
EAL: memzone_reserve_aligned_thread_unsafe(): No more room in config
[2021-06-02 15:08:03.846887] ERROR nvme.c(347) nvme_driver_init: primary process failed to reserve memory
[2021-06-02 15:08:03.846915] ERROR nvme.c(640) spdk_nvme_probe: Create probe context failed
[2021-06-02 15:08:03.846933] WARNING driver.c(1002) nvme_probe: not found device: 0000:02:00.0, rc -1, cb_ctx.ctrlr (nil)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "driver_wrap.pyx", line 562, in nvme.Pcie.__cinit__
nvme.NvmeEnumerateError: fail to create the controller
>>> n=d.Controller(d.Pcie('02:0.0'))
EAL: memzone_reserve_aligned_thread_unsafe(): No more room in config
[2021-06-02 15:08:18.502762] ERROR nvme.c(347) nvme_driver_init: primary process failed to reserve memory
[2021-06-02 15:08:18.502796] ERROR nvme.c(640) spdk_nvme_probe: Create probe context failed
[2021-06-02 15:08:18.502809] WARNING driver.c(1002) nvme_probe: not found device: 02:0.0, rc -1, cb_ctx.ctrlr (nil)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "driver_wrap.pyx", line 562, in nvme.Pcie.__cinit__
nvme.NvmeEnumerateError: fail to create the controller
>>> n=d.Controller(d.Pcie('02:00.0'))
EAL: memzone_reserve_aligned_thread_unsafe(): No more room in config
[2021-06-02 15:08:23.190823] ERROR nvme.c(347) nvme_driver_init: primary process failed to reserve memory
[2021-06-02 15:08:23.190851] ERROR nvme.c(640) spdk_nvme_probe: Create probe context failed
[2021-06-02 15:08:23.190865] WARNING driver.c(1002) nvme_probe: not found device: 0000:02:00.0, rc -1, cb_ctx.ctrlr (nil)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "driver_wrap.pyx", line 562, in nvme.Pcie.__cinit__
nvme.NvmeEnumerateError: fail to create the controller

cranechu, 2021-06-03:
Hi, nate12112, 

Could you please run "make info", and copy the output? Thanks!

nate12112, 2021-06-03:
$ make info
sudo ./src/setup.sh status
Hugepages
node     hugesize     free /  total
node0   1048576kB        0 /      0
node0      2048kB     1087 /   1215

NVMe devices
BDF             Vendor  Device  NUMA    Driver          Device name
0000:03:00.0    1344    5406    -1      uio_pci_generic         -

I/OAT DMA
BDF             Vendor  Device  NUMA    Driver

virtio
BDF             Vendor  Device  NUMA    Driver          Device name
VMD
BDF             Numa Node       Driver Name
sudo cat /proc/meminfo
MemTotal:       15995316 kB
MemFree:         7071832 kB
MemAvailable:   10140432 kB
Buffers:          273908 kB
Cached:          2973996 kB
SwapCached:            0 kB
Active:          3911216 kB
Inactive:        1839928 kB
Active(anon):    2607820 kB
Inactive(anon):    51404 kB
Active(file):    1303396 kB
Inactive(file):  1788524 kB
Unevictable:      103392 kB
Mlocked:               0 kB
SwapTotal:       1044476 kB
SwapFree:        1044476 kB
Dirty:               292 kB
Writeback:             0 kB
AnonPages:       2533368 kB
Mapped:           309904 kB
Shmem:            155992 kB
KReclaimable:     317912 kB
Slab:             391156 kB
SReclaimable:     317912 kB
SUnreclaim:        73244 kB
KernelStack:       21604 kB
PageTables:        86988 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     7797972 kB
Committed_AS:   13080448 kB
VmallocTotal:   34359738367 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
Percpu:             1856 kB
HardwareCorrupted:     0 kB
AnonHugePages:   1722368 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
HugePages_Total:    1215
HugePages_Free:     1087
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:         2488320 kB
DirectMap4k:      205876 kB
DirectMap2M:     4870144 kB
DirectMap1G:    11534336 kB
sudo cat /proc/cpuinfo
processor       : 0
[... 210 lines cut ...]
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

sudo cat /etc/*release
CentOS Linux release 7.7.1908 (Core)
NAME="CentOS Linux"
VERSION="7 (Core)"
ID="centos"
ID_LIKE="rhel fedora"
VERSION_ID="7"
PRETTY_NAME="CentOS Linux 7 (Core)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:centos:centos:7"
HOME_URL="https://www.centos.org/"
BUG_REPORT_URL="https://bugs.centos.org/"

CENTOS_MANTISBT_PROJECT="CentOS-7"
CENTOS_MANTISBT_PROJECT_VERSION="7"
REDHAT_SUPPORT_PRODUCT="centos"
REDHAT_SUPPORT_PRODUCT_VERSION="7"

CentOS Linux release 7.7.1908 (Core)
CentOS Linux release 7.7.1908 (Core)

PM-Zefman-R, 2021-12-22:
I have the same problem on Ubuntu 20.04:
:/usr/local/src/pynvme$ python3
Python 3.8.10 (default, Nov 26 2021, 20:14:08) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> import pytest
>>> import nvme as d
>>> 
>>> pcie = d.Pcie('0000:00:0e.0')
EAL: memzone_reserve_aligned_thread_unsafe(): No more room in config
[2021-12-22 11:09:25.458459] ERROR nvme.c(347) nvme_driver_init: primary process failed to reserve memory
[2021-12-22 11:09:25.458483] ERROR nvme.c(640) spdk_nvme_probe: Create probe context failed
[2021-12-22 11:09:25.458496] WARNING driver.c(1001) nvme_probe: not found device: 0000:00:0e.0, rc -1, cb_ctx.ctrlr (nil)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "driver_wrap.pyx", line 562, in nvme.Pcie.__cinit__
nvme.NvmeEnumerateError: fail to create the controller
>>> pcie = d.Pcie('0000:00:0e.0')

PM-Zefman-R, 2021-12-22:
:/usr/local/src/pynvme$ make info
sudo ./src/setup.sh status
Hugepages
node     hugesize     free /  total
node0      2048kB     1087 /   1215

NVMe devices
BDF		Vendor	Device	NUMA	Driver		Device name
0000:00:0e.0	80ee	4e56	0	uio_pci_generic		-

I/OAT DMA
BDF		Vendor	Device	NUMA	Driver

virtio
BDF		Vendor	Device	NUMA	Driver		Device name
VMD
BDF		Numa Node	Driver Name
sudo cat /proc/meminfo
MemTotal:        8145376 kB
MemFree:          230096 kB
MemAvailable:    3338964 kB
Buffers:          399488 kB
Cached:          2163696 kB
SwapCached:            4 kB
Active:          1400964 kB
Inactive:        2879864 kB
Active(anon):       3604 kB
Inactive(anon):  1838536 kB
Active(file):    1397360 kB
Inactive(file):  1041328 kB
Unevictable:          32 kB
Mlocked:              32 kB
SwapTotal:       2097148 kB
SwapFree:        2095600 kB
Dirty:               252 kB
Writeback:             0 kB
AnonPages:       1717808 kB
Mapped:           571880 kB
Shmem:            128164 kB
KReclaimable:     973892 kB
Slab:            1054076 kB
SReclaimable:     973892 kB
SUnreclaim:        80184 kB
KernelStack:       10640 kB
PageTables:        24048 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     4925676 kB
Committed_AS:    6792848 kB
VmallocTotal:   34359738367 kB
VmallocUsed:       39756 kB
VmallocChunk:          0 kB
Percpu:             2896 kB
HardwareCorrupted:     0 kB
AnonHugePages:         0 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
FileHugePages:         0 kB
FilePmdMapped:         0 kB
HugePages_Total:    1215
HugePages_Free:     1087
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:         2488320 kB
DirectMap4k:      231360 kB
DirectMap2M:     8157184 kB
sudo cat /proc/cpuinfo
processor	: 0
[... 190 lines cut ...]
Ср 22 дек 2021 11:17:11 MSK
pwd
/usr/local/src/pynvme
git status -sb
## master...origin/master
 M .vscode/launch.json
 M .vscode/settings.json
 M driver_test.py
 m spdk
 M src/intr_mgt.c
?? .cproject
?? .project
?? .pydevproject
?? .settings/
?? report.xls
?? scripts/driver_test.py
?? test_init_nvme_customerized-2-report.xls
?? test_init_nvme_customerized-report.xls
git --no-pager log -1
commit 52967ef51dce35cfae196ee312cb81862536e0e2 (HEAD -> master, origin/master, origin/HEAD)
Author: Crane Chu <cranechu@gmail.com>
Date:   Fri Apr 30 21:30:24 2021 +0800

    Add files via upload
:/usr/local/src/pynvme$

PM-Zefman-R, 2021-12-27:
I think, the problem was in debug settings - [launch.json]:
"
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Python: Current File",
            "type": "python",
            "request": "launch",
            "program": "${file}",
            "console": "integratedTerminal",
            "justMyCode": false
        },
        {
            "name": "pytest case",
            "type": "python",
            "request": "test",
            "console": "integratedTerminal",
            "sudo": true,
            "justMyCode": false
        },
        {
            "name": "python file",
            "type": "python",
            "request": "launch",
            "program": "${file}",
            "console": "integratedTerminal",
            "sudo": true,
            "justMyCode": false
        }
    ]
}
"
With this the problem disappears.

closed 2022-10-07 by cranechu