pynvme/pynvme #107: Failed install due to Python requirements

cyamato, 2022-01-14:
OS: Ubuntu 20.04.3
Python: 3.10.1+

Error: error in pyserial setup command: use_2to3 is invalid.
Source: pyserial==2.7 in requirments.txt
When: During install with ./install.sh and with python3 -m pip -r requirments.txt

Fix: Migrate to pyserial~=3.0

Documentation should include Python verison requirements and or versions tested under.

cranechu, 2022-01-18:
Please try to install pynvme on the fresh installed Ubuntu 20.04.3. If you meet any problem, then we can reproduce it quickly. 
For example, the version of Python3 in my Ubuntu 20.04.3 is 3.8.10.

cyamato, 2022-01-18:
Failed on clean bare metal install with Python 3.10.1+ with the same error code related to using of 2to3.
Sucess if using an older version of Python or updating the required version of pyserial.

Error is the result of the early depreciation of the lib2to3 module in python 3.10.

Please see https://github.com/pypa/setuptools/issues/2086 for more details.

Can pyserial be revved up to > 3.0 which does not require 2to3?  If a Conda Python 3.8 version is used to install would there be any issues with using it in a Python 3.10+ environment?

closed 2022-10-07 by cranechu