pynvme/pynvme #109: pcie.cap_offset(0xC) always returns None (PCI Link Capabilities register)

dilanpace1496, 2022-01-17:
Trying to find ASPM capability of the Link Capabilities register (not ASPM of Link Control Register) however pcie.cap_offset(0xC) will always return None.

Running an lspci -vv -s ##:##.## shows that the PCI Link Capabilities register exists and contains the ASPM field I am looking to check.

I have tried multiple drive samples as well as a few host machines but have not gotten an offset of the register returned and only 'None'.

cranechu, 2022-01-18:
the parameter of pcie.cap_offset() is the Capability ID, and the capability id of PCI Express Capability is 0x10, instead of 0xC. 0xC is the offset of PCI Express Link Capabilities register in PCI Express Capability.

dilanpace1496, 2022-01-18:
Thanks for the help, new to the realm of the PCIe specification but that worked.

closed 2022-01-18 by dilanpace1496