pynvme/pynvme #110: How to test one device that has multiple controllers and multiple namespaces (multiple physical function)?

VincentLi212, 2022-01-21:
Hi, 
Who has tested one device that has multiple controllers and multiple namespaces (multiple physical function)?

cranechu, 2022-01-21:
https://github.com/pynvme/pynvme/blob/52967ef51dce35cfae196ee312cb81862536e0e2/scripts/test_examples.py#L149

You can refer to the example above. 

Thanks, 
-Crane

cranechu, 2022-01-22:
Hi, Vincent, 

1. can you share the script code of creating ns_list and its elements?
2. you can print and check the namespace size by: logging.info(ns_list[i].id_data(7, 0))

Thanks, 
-Crane

VincentLi212, 2022-01-23:
Hi Crane,
Could you give me some ideas?

I print and check the namespace size, size of NSID 1 is 465567744, size of other NSID is 0
I use "sudo fdisk -l" can find all namespace size is "Disk /dev/nvme4n1: 222 GiB, 238370684928 bytes, 465567744 sectors"
[2022-01-23 19:05:02.042] INFO test_mix_io(66): ns_list 0.id_data(7, 0) =  465567744
[2022-01-23 19:05:02.042] INFO test_mix_io(66): ns_list 1.id_data(7, 0) =  0
[2022-01-23 19:05:02.042] INFO test_mix_io(66): ns_list 2.id_data(7, 0) =  0
[2022-01-23 19:05:02.042] INFO test_mix_io(66): ns_list 3.id_data(7, 0) =  0
[2022-01-23 19:05:02.042] INFO test_mix_io(66): ns_list 4.id_data(7, 0) =  0
[2022-01-23 19:05:02.042] INFO test_mix_io(66): ns_list 5.id_data(7, 0) =  0
[2022-01-23 19:05:02.042] INFO test_mix_io(66): ns_list 6.id_data(7, 0) =  0
[2022-01-23 19:05:02.042] INFO test_mix_io(66): ns_list 7.id_data(7, 0) =  0

Below is my script code of creating ns_list.
def get_device():
    pcie_addr_list = get_pcie_addr()
    pcie_list = [d.Pcie(b) for b in pcie_addr_list]
    nvme_list = [d.Controller(a) for a in pcie_list]
    ns_list = [d.Namespace(n) for n in nvme_list]
    return pcie_list, nvme_list, ns_list

def get_pcie_addr():
    '''
    Get pcie address from "lspci" command
    '''
    pcie_addr_list = []
    cmd = "lspci | grep Technology"   # key word of driver
    ret, out = execute_cmd(cmd, 10, out_flag = True)
    if out == []:
        logging.error("No device found")
    else:
        for item in out:
            pcie_addr = item.split(" ")[0].strip()
        pcie_addr_list.append(pcie_addr)
        logging.info("pcie_addr list: {}".format(pcie_addr_list))
    return pcie_addr_list

Below is nvme_list and ns_list log:
[2022-01-23 19:04:56.034] INFO test_mix_io(56): nvme_list [<nvme.Controller object at 0x7ffff31f8e50>, <nvme.Controller object at 0x7ffff2fad770>, <nvme.Controller object at 0x7ffff2fb2090>, <nvme.Controller object at 0x7ffff2fb2950>, <nvme.Controller object at 0x7ffff2fb6270>, <nvme.Controller object at 0x7ffff2fb6b30>, <nvme.Controller object at 0x7ffff2fbb450>, <nvme.Controller object at 0x7ffff2fbbd10>]

[2022-01-23 19:04:56.034] INFO test_mix_io(57): ns_list [<nvme.Namespace object at 0x7ffff2fbe630>, <nvme.Namespace object at 0x7ffff2fbed60>, <nvme.Namespace object at 0x7ffff2f522c0>, <nvme.Namespace object at 0x7ffff2f52590>, <nvme.Namespace object at 0x7ffff2f52860>, <nvme.Namespace object at 0x7ffff2f52b30>, <nvme.Namespace object at 0x7ffff2f52e00>, <nvme.Namespace object at 0x7ffff2f57130>]

Thanks,
Vincent

cranechu, 2022-01-24:
| I print and check the namespace size, size of NSID 1 is 465567744, size of other NSID is 0
I think this caused the problem you met. You can double check the namspace identify data of other namespacs first.

VincentLi212, 2022-01-24:
I'm confused that FIO can test all namespaces and see that all namespaces have the same capacity by command "sudo fdisk -l". When Pynvme takes over,  size of other namespace is 0 except first namespace.

cranechu, 2022-01-24:
It's quite usual.  :) I suggest you verify it with your firmware engineers first. 
Is the namespace's nsid is 1 in all other controller?

VincentLi212, 2022-01-25:
Hi Crane,

I have verified with our firmware engineer, and I use nvme-cli to get the nsid of all namespace, NSID is  1 2 3 4 5 6 7 8.
Pynvme version is 2.3.2

Here is one subsystem, this subsystem has  8 controllers and every controller has one namespace.
I find pynvme init nvme0 with NSID 1, nvme1 with NSID 2, nvme2 with NSID 3 ......... (second log)
After nvme init, I find pynvme always init NSID 1. but I think it shall init NSID 1 2 3 4 5 6 7 8.  (third log)
Because here is one subsystem, this subsystem has  8 controllers and every controller has one namespace.

Below is nvme-cli log:
user@gen4-06:~/pynvme$ sudo nvme list-ns /dev/nvme0n1
[   0]:0x1
user@gen4-06:~/pynvme$ sudo nvme list-ns /dev/nvme1n1
[   0]:0x2
user@gen4-06:~/pynvme$ sudo nvme list-ns /dev/nvme2n1
[   0]:0x3
user@gen4-06:~/pynvme$ sudo nvme list-ns /dev/nvme3n1
[   0]:0x4
user@gen4-06:~/pynvme$ sudo nvme list-ns /dev/nvme4n1
[   0]:0x5
user@gen4-06:~/pynvme$ sudo nvme list-ns /dev/nvme5n1
[   0]:0x6
user@gen4-06:~/pynvme$ sudo nvme list-ns /dev/nvme6n1
[   0]:0x7
user@gen4-06:~/pynvme$ sudo nvme list-ns /dev/nvme7n1
[   0]:0x8

Get NSID 2 successfully
[2022-01-25 14:55:42.237] DEBUG <listcomp>(16): nvme initialized: b'0000:07:00.1'
[2022-01-25 14:55:42.237] DEBUG <listcomp>(16): start nvme init process in pynvme
[2022-01-25 14:55:42.239] DEBUG <listcomp>(16): send admin command, opcode 6h
[2022-01-25 14:55:42.239] DEBUG <listcomp>(16): to reap 1 admin commands
[2022-01-25 14:55:42.240267] NOTICE nvme_qpair.c(119) nvme_admin_qpair_print_command: IDENTIFY (06) sqid:0 cid:126 nsid:1 cdw10:00000003 cdw11:00000000
[2022-01-25 14:55:42.240280] NOTICE nvme_qpair.c(315) spdk_nvme_qpair_print_completion: INVALID FIELD (00/02) sqid:0 cid:126 cdw0:0 sqhd:0003 p:1 m:0 dnr:1
[2022-01-25 14:55:42.240284] WARNING nvme_ns.c(161) nvme_ctrlr_identify_id_desc: Failed to retrieve NS ID Descriptor List
[2022-01-25 14:55:42.240753] NOTICE nvme_qpair.c(119) nvme_admin_qpair_print_command: IDENTIFY (06) sqid:0 cid:126 nsid:3 cdw10:00000003 cdw11:00000000
[2022-01-25 14:55:42.240756] NOTICE nvme_qpair.c(315) spdk_nvme_qpair_print_completion: INVALID FIELD (00/02) sqid:0 cid:126 cdw0:0 sqhd:0007 p:1 m:0 dnr:1
[2022-01-25 14:55:42.240759] WARNING nvme_ns.c(161) nvme_ctrlr_identify_id_desc: Failed to retrieve NS ID Descriptor List
[2022-01-25 14:55:42.240975] NOTICE nvme_qpair.c(119) nvme_admin_qpair_print_command: IDENTIFY (06) sqid:0 cid:126 nsid:4 cdw10:00000003 cdw11:00000000

[2022-01-25 14:55:42.431] DEBUG <listcomp>(17): initialize namespace nsid 1
[2022-01-25 14:55:42.730] DEBUG <listcomp>(17): initialize namespace nsid 1
[2022-01-25 14:55:42.731] DEBUG <listcomp>(17): initialize namespace nsid 1
[2022-01-25 14:55:42.731] DEBUG <listcomp>(17): initialize namespace nsid 1
[2022-01-25 14:55:42.731] DEBUG <listcomp>(17): initialize namespace nsid 1
[2022-01-25 14:55:42.731] DEBUG <listcomp>(17): initialize namespace nsid 1
[2022-01-25 14:55:42.731] DEBUG <listcomp>(17): initialize namespace nsid 1
[2022-01-25 14:55:42.731] DEBUG <listcomp>(17): initialize namespace nsid 1

Thanks
Vincent

cranechu, 2022-01-25:
https://pynvme.readthedocs.io/en/latest/api.html#namespace

The default nsid parameter of Namespace is 1, but you can specify different value (e.g. 2-8). 
You can try to fix it in the line below: 
> nvme_list = [d.Controller(a) for a in pcie_list]

cranechu, 2022-01-25:
You can also consider commercial product PyNVMe3 (https://pynv.me/3/) for more features and better support!  :)

VincentLi212, 2022-01-27:
Hi Crane,

Thanks for your help, I specified different value and fix it.
As for PyNVMe3, our team will consider it in the future.
Thanks a lot.

Thanks
Vincent

closed 2022-01-27 by VincentLi212