pynvme/pynvme #111: Supporting MSI mode

williszheng, 2022-02-16:
Hi:
Does pynvme support MSI interrupt mode ？ 
From /src/intr_mgt.c we found that: 
void intc_init(struct spdk_nvme_ctrlr *ctrlr)
{
  bool ret;
  
  // interrupt is enabled on PCIe devices only
  assert(ctrlr->trid.trtype == SPDK_NVME_TRANSPORT_PCIE);

  //search msix first, if the operation fail, will switch to msi intr
  ret = msix_intc_init(ctrlr, &ctrlr->pynvme_intc_ctrl);
  if (ret == false)
  {
    ret = msi_intc_init(ctrlr, &ctrlr->pynvme_intc_ctrl);
  }

  assert(ret == true);  // controller must support at least one kind of interrupt
}

Does THE msi_intc_init function be work normally ？

Waiting for your answer and appreciate your help

cranechu, 2022-02-17:
Hi, pynvme supports both MSIx and MSI.

williszheng, 2022-02-28:
Is there any tip about how to enable it ?  :v

cranechu, 2022-02-28:
If the DUT does not support MSIx, but supports MSI, MSI will be enabled automatically.

OUZYcn, 2022-06-13:
https://pynvme.readthedocs.io/en/latest/intro.html

please update document. 
[image]

closed 2022-10-07 by cranechu