pynvme/pynvme #112: Problem installing pynvme

idonttest, 2022-06-30:
Hello, 
I have been trying to install pynvme using pip. 
```

"pip install pynvme"
"cd "~/.local/pynvme"
"make setup "
"make test"

```
upon one fresh install try i can only "import pynvme" but it has no sub commands. 
sanother fresh install  i can call "import nvme" and i get 
`ImportError: /home/test/.local/pynvme/nvme.so: undefined symbol: _Py_CheckRecursionLimit`   This appears to be a problem with python version and deprecated fn. 

Python version: 3.10 
OS: Ubuntu 20.02

I did try to compile from source and it resulted in more rabbit holes than this method.

Am i missing something??

cranechu, 2022-07-01:
Please follow this page to install: https://pynvme.readthedocs.io/en/latest/install.html

idonttest, 2022-07-01:
Thanks for the quick response
I did try that and it has a compilation in dpdk. 
below is the log of install.shh

`
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
python3-dev is already the newest version (3.10.4-0ubuntu2).
python3-setuptools is already the newest version (59.6.0-1.2).
python3-tk is already the newest version (3.10.4-0ubuntu1).
python3-pip is already the newest version (22.0.2+dfsg-1).
0 upgraded, 0 newly installed, 0 to remove and 183 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
g++ is already the newest version (4:11.2.0-1ubuntu1).
gcc is already the newest version (4:11.2.0-1ubuntu1).
libaio-dev is already the newest version (0.3.112-13build1).
libiscsi-dev is already the newest version (1.19.0-3build2).
make is already the newest version (4.3-4.1build1).
pciutils is already the newest version (1:3.7.0-6).
sg3-utils is already the newest version (1.46-1build1).
uuid-dev is already the newest version (2.37.2-4ubuntu3).
astyle is already the newest version (3.1-2build1).
clang is already the newest version (1:14.0-55~exp2).
lcov is already the newest version (1.15-1).
libcunit1-dev is already the newest version (2.1-3-dfsg-2.4).
pep8 is already the newest version (1.7.1-9ubuntu1).
shellcheck is already the newest version (0.8.0-2).
git is already the newest version (1:2.34.1-1ubuntu1.2).
libssl-dev is already the newest version (3.0.2-0ubuntu1.5).
0 upgraded, 0 newly installed, 0 to remove and 183 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
pycodestyle is already the newest version (2.8.0-2).
0 upgraded, 0 newly installed, 0 to remove and 183 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
libunwind-dev is already the newest version (1.3.2-2build2).
0 upgraded, 0 newly installed, 0 to remove and 183 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
libibverbs-dev is already the newest version (39.0-1).
librdmacm-dev is already the newest version (39.0-1).
0 upgraded, 0 newly installed, 0 to remove and 183 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
libnuma-dev is already the newest version (2.0.14-3ubuntu2).
nasm is already the newest version (2.15.05-1).
0 upgraded, 0 newly installed, 0 to remove and 183 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
doxygen is already the newest version (1.9.1-2ubuntu2).
graphviz is already the newest version (2.42.2-6).
mscgen is already the newest version (0.20-14).
0 upgraded, 0 newly installed, 0 to remove and 183 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
python3-configshell-fb is already the newest version (1:1.1.28-2).
python3-pexpect is already the newest version (4.8.0-2ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 183 not upgraded.
[... 1432 lines cut ...]
ff:1e.1 System peripheral: Intel Corporation Xeon E7 v4/Xeon E5 v4/Xeon E3 v4/Xeon D Power Control Unit (rev 01)
ff:1e.2 System peripheral: Intel Corporation Xeon E7 v4/Xeon E5 v4/Xeon E3 v4/Xeon D Power Control Unit (rev 01)
ff:1e.3 System peripheral: Intel Corporation Xeon E7 v4/Xeon E5 v4/Xeon E3 v4/Xeon D Power Control Unit (rev 01)
ff:1e.4 System peripheral: Intel Corporation Xeon E7 v4/Xeon E5 v4/Xeon E3 v4/Xeon D Power Control Unit (rev 01)
ff:1f.0 System peripheral: Intel Corporation Xeon E7 v4/Xeon E5 v4/Xeon E3 v4/Xeon D Power Control Unit (rev 01)
ff:1f.2 System peripheral: Intel Corporation Xeon E7 v4/Xeon E5 v4/Xeon E3 v4/Xeon D Power Control Unit (rev 01)
date
vr  1 jul 2022 13:37:22 CEST
pwd
/home/testy/pyy/1/pynvme
git status -sb
## master...origin/master
 M .vscode/settings.json
?? allout.txt
?? install_log
git --no-pager log -1
commit 52967ef51dce35cfae196ee312cb81862536e0e2
Author: Crane Chu <cranechu@gmail.com>
Date:   Fri Apr 30 21:30:24 2021 +0800

    Add files via upload
sudo python3 -B -m pytest scripts/test_examples.py::test_hello_world --pciaddr= -s -v -r Efsx --excelreport=report.xls --verbose
make[1]: Leaving directory '/home/testy/pyy/1/pynvme'
sudo rm -rf .pytest_cache
pynvme install done.`

idonttest, 2022-07-01:
Thanks for the quick response
I did try that and it has a compilation in dpdk.
below is the log of install.shh
[allout.txt](https://github.com/pynvme/pynvme/files/9028539/allout.txt)

cranechu, 2022-07-01:
Which OS/version you are using? The open source pynvme is only tested with Fedora 32, so it may need some fixes to work with different OS/version.

catogts, 2022-07-02:
I tested in fedora 32,34,36 
32,34 no need additional action
but 36 need some fix in md5 header file in fedora 36

closed 2022-10-07 by cranechu