pynvme/pynvme #118: Ubuntu 20.04.1安装pynvme失败

yxjhy001, 2023-12-08:
CC module/bdev/lvol/vbdev_lvol.o
  CC module/bdev/error/vbdev_error.o
  CC module/bdev/null/bdev_null.o
  CC module/bdev/nvme/bdev_nvme.o
  CC module/bdev/gpt/vbdev_gpt.o
  CC module/blobfs/bdev/blobfs_bdev_rpc.o
  CC module/bdev/error/vbdev_error_rpc.o
  CC module/bdev/null/bdev_null_rpc.o
  CC module/bdev/malloc/bdev_malloc_rpc.o

make[1]: Entering directory '/root/yxj/pynvme/src'
  CC src/driver.o
  CC src/intr_mgt.o
  CC src/ioworker.o
  LIB libspdk_pynvme.a
Compiling driver_wrap.pyx because it changed.
[1/1] Cythonizing driver_wrap.pyx

Error compiling Cython file:
------------------------------------------------------------
...
        self._timeout = _cTIMEOUT*1000
        self.nvme_init_func = nvme_init_func
        self.aer_cb_func = None

        # register timeout callback
        d.nvme_register_timeout_cb(self.pcie._ctrlr, timeout_driver_cb, self._timeout)
                                                     ^
------------------------------------------------------------

driver_wrap.pyx:900:53: Cannot assign type 'void (void *, ctrlr *, qpair *, unsigned short) except *' to 'timeout_cb_func'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (void *, ctrlr *, qpair *, unsigned short) except *'.

Error compiling Cython file:
------------------------------------------------------------
...
        # Parameters
            msec (int): milli-seconds of timeout value
        """

        self._timeout = msec
        d.nvme_register_timeout_cb(self.pcie._ctrlr, timeout_driver_cb, self._timeout)
                                                     ^
------------------------------------------------------------

driver_wrap.pyx:1026:53: Cannot assign type 'void (void *, ctrlr *, qpair *, unsigned short) except *' to 'timeout_cb_func'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (void *, ctrlr *, qpair *, unsigned short) except *'.

Error compiling Cython file:
------------------------------------------------------------
...
                            cdw11=0,
                            cdw12=0,
                            cdw13=0,
                            cdw14=0,
                            cdw15=0,
                            cb_func=cmd_cb,
                                    ^
------------------------------------------------------------

driver_wrap.pyx:1200:36: Cannot assign type 'void (void *, const cpl *) except *' to 'cmd_cb_func'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (void *, const cpl *) except *'.

Error compiling Cython file:
------------------------------------------------------------
...
                            cdw11=(csi << 24) | nvmsetid,
                            cdw12=0,
                            cdw13=0,
                            cdw14=0,
                            cdw15=0,
                            cb_func=cmd_cb,
                                    ^
[... 390 lines cut ...]
make[1]: Entering directory '/root/yxj/pynvme'

root@alibaba-System-Product-Name:~/yxj/pynvme# make setup
sudo sh -c "echo 1 > /sys/bus/pci/rescan"
sudo rm -rf /run/dpdk
sudo rm -rf /var/run/dpdk
sudo ./src/setup.sh cleanup
ls: cannot access '/dev/shm/*': No such file or directory
Clean
sudo ./src/setup.sh reset
0000:01:00.0 (1ded 5236): uio_pci_generic -> nvme
sudo rm -f /var/tmp/spdk.sock*
sudo rm -f /var/tmp/pynvme.sock*
sudo rm -rf .pytest_cache
sudo fuser -k 4420/tcp
make: [Makefile:73: reset] Error 1 (ignored)
sudo sh -c 'find . | grep -E "(__pycache__|\.pyc|\.pyo$)" | xargs rm -rf'
xhost +local:           # enable GUI with root/sudo
xhost:  unable to open display ""
make: [Makefile:93: setup] Error 1 (ignored)
sudo chmod 777 /tmp
sudo sh -c 'find . | grep -E "(__pycache__|\.pyc|\.pyo$)" | xargs rm -rf'
sed -i 's/XXXX:BB:DD.F/0000:01:00.0/g' .vscode/settings.json
sudo HUGEMEM=2430    DRIVER_OVERRIDE=uio_pci_generic ./src/setup.sh     # UIO is recommended
0000:01:00.0 (1ded 5236): nvme -> uio_pci_generic

Harish888-404, 2024-02-07:
Could you please let me know how to fix this issue? I'm on the same ubuntu version

Gracecao199312, 2024-08-21:
it seems your python lib cython is not matched, assignment other cython version is a smart way:pip3 install Cython==3.0.0a10